google-deepmind/mujoco #1138: GLXBadContext when using mujoco.viewer.launch_passive()

manavkulshrestha, 2023-10-26:
Hi, I'm running into this error when running episodes with the viewer

```
X Error of failed request:  GLXBadContext
  Major opcode of failed request:  152 (GLX)
  Minor opcode of failed request:  4 (X_GLXDestroyContext)
  Serial number of failed request:  355
  Current serial number in output stream:  355
X Error of failed request:  GLXBadContext
  Major opcode of failed request:  152 (GLX)
  Minor opcode of failed request:  4 (X_GLXDestroyContext)
  Serial number of failed request:  355
  Current serial number in output stream:  355
```
I'm not entirely sure what could be causing this.
My code for launching the viewer is as follows:
```
import mujoco as mj
import mujoco.viewer as mjv

def run_episode(path, episode_length=2000):
  m = mj.MjModel.from_xml_path(str(path))
  d = mj.MjData(m)

  t = 0
  data = []
  
  with mjv.launch_passive(model, data) as viewer:
    while viewer.is_running() and t < episode_length:
      # policy code omitted
  
      mj.mj_step(m, d)
      viewer.sync()
  
      # data collection code. collect omitted
      data.append(collect(m, d))
      
      t += 1
      time.sleep(0.01)
      
  return data

if __name__ == '__main__':
  run_episode(sys.argv[1])
```
I have not included the mjcf, but can provide it if that may be contributing to the issue.

- Include the following context:
  - Ubunutu 22.04
  - Python 3.10.13
  - mujoco.mj_version() returns 300
  - Using official python bindings

assadollahi, 2023-11-24:
have same problem.

`import time

import mujoco
import mujoco.viewer

global angle

def key_callback(keycode):
  global angle
  if chr(keycode) == '1':
    angle += 0.1

m = mujoco.MjModel.from_xml_path('./kayraCircle.xml')
d = mujoco.MjData(m)
angle = 0

#with mujoco.viewer.launch_passive(m, d) as viewer:
with mujoco.viewer.launch_passive(m, d, key_callback=key_callback) as viewer:
  # Close the viewer automatically after 30 wall-seconds.
  start = time.time()

  while viewer.is_running() and time.time() - start < 30:
    step_start = time.time()

    # mj_step can be replaced with code that also evaluates
    # a policy and applies a control signal before stepping the physics.
    mujoco.mj_step(m, d)

    # Example modification of a viewer option: toggle contact points every two seconds.
    with viewer.lock():
      viewer.opt.flags[mujoco.mjtVisFlag.mjVIS_CONTACTPOINT] = int(d.time % 2)
      d.actuator("Right Calf 3BR").ctrl = 1.57 - angle

    # Pick up changes to the physics state, apply perturbations, update options from GUI.
    viewer.sync()

    # Rudimentary time keeping, will drift relative to wall clock.
    time_until_next_step = m.opt.timestep - (time.time() - step_start)
    if time_until_next_step > 0:
      time.sleep(time_until_next_step)`

> X Error of failed request:  GLXBadContext
  Major opcode of failed request:  152 (GLX)
  Minor opcode of failed request:  4 (X_GLXDestroyContext)
  Serial number of failed request:  1042
  Current serial number in output stream:  1042
X Error of failed request:  GLXBadContext
  Major opcode of failed request:  152 (GLX)
  Minor opcode of failed request:  4 (X_GLXDestroyContext)
  Serial number of failed request:  1042
  Current serial number in output stream:  1042

kevinzakka, 2023-12-03:
What graphics card do you have out of curiosity?

assadollahi, 2023-12-04:
I have an NVIDIA RTX 3090, also running Ubuntu 22.04

saran-t, 2024-02-01:
Sorry for the late reply, I just returned to work. Can you identify which line of code is triggering the error?

kevinzakka, 2024-08-30:
Going to close due to inactivity. Please re-open if this is still an issue.

closed 2024-08-30 by kevinzakka