google-deepmind/mujoco #1139: XML Error: unknown plugin

xnliu91, 2023-10-26:
Hi All,

Our team is working with build-from-source Mujooco. I just pulled the latest Mujoco repo on my dev machine (Ubuntu 22.04). I found the plugins are broken again when loading the xml examples under share/mujoco/model/plugin, e.g.
`XML Error: unknown plugin 'mujoco.elasticity.cable'
Element 'plugin', line 20`

I used to be able to run them with this fix: https://github.com/google-deepmind/mujoco/issues/723 but after pulling it's not working anymore.

Can you please help with this issue? Thanks!

saran-t, 2023-10-26:
are there plugin files in the `mujoco_plugin` directory?

xnliu91, 2023-10-26:
@saran-t yes Nvm I figured it out.

Instead of putting the libraries under bin/mujoco_plugin which worked in the previous version, now mujoco_plugin need to be under project root.

Thanks! If you can fix CMake/build from source documentation that would be helpful! Or I can send a pr once I have time :)

saran-t, 2023-10-26:
That shouldn't be the case.. we didn't change our `simulate` loading logic. The `mujoco_plugin` directory just has to be in the same place wherever `simulate` happens to be.

xnliu91, 2023-10-27:
@saran-t actually you're right. I deleted everything and put `mujoco_plugin` folder next to the `simulate` and it worked.

I might have some weird local problems. Sorry for the confusion! I've closed this issue.

Wangzihao12ous, 2024-03-08:
> That shouldn't be the case.. we didn't change our `simulate` loading logic. The `mujoco_plugin` directory just has to be in the same place wherever `simulate` happens to be.

> are there plugin files in the `mujoco_plugin` directory?

Hey! I used this code "m = mj_loadXML("../xml/cable.xml", NULL, error, 1000)" to load model, and the erro is "XML Error: unknown plugin 'mujoco.elasticity.cable'".But if I use _simulate_ to load this model, it is work. Do you know why this error occurred?