google-deepmind/mujoco #1141: Example for deformable objects not working

nico-bohlinger, 2023-10-27:
Hey everyone,

I wanted to try the example for deformable objects from the [documentation](https://mujoco.readthedocs.io/en/latest/modeling.html#deformable-objects) but when using the provided XML in "simulate" I get a warning in the terminal and the model seems to reload after a second or so.

The XML from the documentation:
```xml
<mujoco>
  <extension>
    <plugin plugin="mujoco.elasticity.solid"/>
  </extension>
  <worldbody>
    <flexcomp type="grid" count="24 4 4" spacing=".1 .1 .1" pos=".1 0 1.5" radius=".0" rgba="0 .7 .7 1" name="softbody" dim="3" mass="7">
      <contact condim="3" solref="0.01 1" solimp=".95 .99 .0001" selfcollide="none"/>
      <edge damping="1"/>
      <plugin plugin="mujoco.elasticity.solid">
        <config key="poisson" value="0.2"/>
        <config key="young" value="5e4"/>
      </plugin>
    </flexcomp>
  </worldbody>
</mujoco>
```

The warning in the terminal that I opened "simulate" with:
`WARNING: Nan, Inf or huge value in QACC at DOF 399. The simulation is unstable. Time = 0.1340.`

MuJoCo Version: Release 3.0.0 pre-built binary

quagla, 2023-10-30:
Could you try with 

`<option timestep=".001"/>`

just to confirm that's the issue. If so, we'll add it to the docs.

nico-bohlinger, 2023-10-31:
Yes, it works now!
Maybe one could also add a floor to the example such you can actually see the block deforming and bouncing

commit 7a7cfc0f (2023-11-02): Improve flex docs.

Fixes #1141 and #1152.

PiperOrigin-RevId: 578855375
Change-Id: I44746c3ae331edcb22b7e16744fb38369039e888
closed 2023-11-02 by copybara-service[bot]