google-deepmind/mujoco #1144: GUI graphics is not displayed properly

yzd0014, 2023-10-28:
[image]

After I opened mujoco default visualization tool, somehow some text was cut off. I reinstalled the driver, which doesn't fix the issue. I guess it may be related the graphics library that mujoco, maybe I didn't install the necessary library? Any help will be appreciated, thanks!

yuvaltassa, 2023-10-28:
Do you have an AMD graphics card?
If yes, search the issues for AMD and subscribe to the relevant one. We are working on a fix.

If not, please give more details regarding your setup.

saran-t, 2023-10-28:
The one bit of good news is that your screenshot is pretty much exactly what I saw last week on our AMD machine (7940HS, Adrenalin 23.10.2 driver, Windows 11 Pro), so it's the most likely one to get fixed. 

Otherwise I'll close this as a duplicate of #639. Let's continue discussions there.

closed 2023-10-28 by saran-t

yzd0014, 2023-10-28:
> Do you have an AMD graphics card? If yes, search the issues for AMD and subscribe to the relevant one. We are working on a fix.
> 
> If not, please give more details regarding your setup.

Yes, it's an AMD gpu

yzd0014, 2023-10-28:
> The one bit of good news is that your screenshot is pretty much exactly what I saw last week on our AMD machine (7940HS, Adrenalin 23.10.2 driver, Windows 11 Pro), so it's the most likely one to get fixed.
> 
> Otherwise I'll close this as a duplicate of #639. Let's continue discussions there.

Thanks for confirming that it's an issue related to AMD gpu.