google-deepmind/mujoco #1145: Mismatches between SensorReading and MjHingeJoint Velocity in Unity

anayebi, 2023-10-29:
I imported the 3-link swimmer agent into Unity via the Unity Mujoco Plugin version 2.3.7, and I noticed a mismatch between the `SensorReading` attributes of the velocity sensors and the `Velocity` attributes of their corresponding `MjHingeJoint`.
I am running this on Mac OS X Venture 13.2.1.

In particular, after importing the swimmer and pressing play in Unity, when I print out the `head/segment_0/joint_0`'s `Velocity` attribute it is -0.0002526487, but the `sensors/gyro_0`'s `SensorReading` attribute is `(0, 0, 0)`. 

In particular, I would have expected the `y` (2nd) component of the `sensors/gyro_0`'s `SensorReading` vector to be the same as the `head/segment_0/joint_0`'s `Velocity` attribute as this represents rotational velocity of the first link in the original Mujoco `z`-axis direction (due to swizzling when importing into Unity since the `y`-axis is up in Unity). 

Or am I misunderstanding something and I shouldn't expect these two quantities to be the same?
My main intention is to replicate the observations of the original `DMControl` agent, linked [here](https://github.com/google-deepmind/dm_control/blob/87e046bfeab1d6c1ffb40f9ee2a7459a38778c74/dm_control/suite/swimmer.py#L153-L154).

For reference, here is the XML config of the 3-link swimmer that I am importing into Unity:
```
<mujoco model="swimmer">
  <include file="./common/visual.xml"/>
  <include file="./common/skybox.xml"/>
  <include file="./common/materials.xml"/>

  <option timestep="0.002" density="3000">
    <flag contact="disable"/>
  </option>

  <default>
    <default class="swimmer">
      <joint type="hinge" pos="0 -.05 0" axis="0 0 1" limited="true" solreflimit=".05 1" solimplimit="0 .8 .1" armature="1e-6"/>
      <default class="inertial">
        <geom type="box" size=".001 .05 .01" rgba="0 0 0 0" mass=".01"/>
      </default>
      <default class="visual">
        <geom type="capsule" size=".01" fromto="0 -.05 0 0 .05 0" material="self" mass="0"/>
      </default>
      <site size=".01" rgba="0 0 0 0"/>
    </default>
    <default class="free">
      <joint limited="false" stiffness="0" armature="0"/>
    </default>
    <motor gear="5e-4" ctrllimited="true" ctrlrange="-1 1"/>
  </default>

  <worldbody>
    <geom name="ground" type="plane" size="2 2 0.1" material="grid"/>
    <body name="head" pos="0 0 .05" childclass="swimmer">
      <light name="light_1" diffuse=".8 .8 .8" pos="0 0 1.5"/>
      <geom name="head" type="ellipsoid" size=".02 .04 .017" pos="0 -.022 0" material="self" mass="0"/>
      <geom name="nose" type="sphere" pos="0 -.06 0" size=".004" material="effector" mass="0"/>
      <geom name="eyes" type="capsule" fromto="-.006 -.054 .005 .006 -.054 .005" size=".004" material="eye" mass="0"/>
      <camera name="tracking1" pos="0.0 -0.1 0.25" xyaxes="1 0 0 0 1 1" mode="trackcom" fovy="60"/>
      <camera name="tracking2" pos="-0.45 0.25 0.075" xyaxes="0 -1 0 .3 0 1" mode="trackcom" fovy="60"/>
      <camera name="eyes" pos="0 -.058 .005" xyaxes="-1 0 0 0 0 1"/>
      <joint name="rootx" class="free" type="slide" axis="1 0 0" pos="0 -.05 0"/>
      <joint name="rooty" class="free" type="slide" axis="0 1 0" pos="0 -.05 0"/>
      <joint name="rootz" class="free" type="hinge" axis="0 0 1" pos="0 -.05 0"/>
      <geom name="inertial" class="inertial"/>
      <geom name="visual" class="visual"/>
      <site name="head"/>
    <body name="segment_0" pos="0 .1 0"><geom class="visual" name="visual_0"/><geom class="inertial" name="inertial_0"/><site name="site_0"/><joint name="joint_0" range="-120.0 120.0"/><body name="segment_1" pos="0 .1 0"><geom class="visual" name="visual_1"/><geom class="inertial" name="inertial_1"/><site name="site_1"/><joint name="joint_1" range="-120.0 120.0"/></body></body></body>
    <geom name="target" type="sphere" pos="1 1 .05" size=".1" material="target"/>
    <light name="target_light" diffuse="1 1 1" pos="1 1 1.5"/>
  </worldbody>

  <sensor>
    <framepos name="nose_pos" objtype="geom" objname="nose"/>
    <framepos name="target_pos" objtype="geom" objname="target"/>
    <framexaxis name="head_xaxis" objtype="xbody" objname="head"/>
    <frameyaxis name="head_yaxis" objtype="xbody" objname="head"/>
    <velocimeter name="head_vel" site="head"/>
    <gyro name="head_gyro" site="head"/>
  </sensor>

<actuator><motor name="motor_0" joint="joint_0"/><motor name="motor_1" joint="joint_1"/></actuator><sensor><velocimeter name="velocimeter_0" site="site_0"/><gyro name="gyro_0" site="site_0"/><velocimeter name="velocimeter_1" site="site_1"/><gyro name="gyro_1" site="site_1"/></sensor></mujoco>
```

Balint-H, 2023-10-30:
The SensorReading property is nonzero for me with the same model, also on 2.3.7 . Do you have any exceptions shown in the console indicating the MuJoCo scene could not be initialized? Also, when in the update loop are you getting your SensorReading?

Lastly I'd definitely expect some discrepancy between a gyro reading and a joint's qvel, as the joint qvel is giving you local angular velocity frame, while (as I understand it) the gyro is global angular acceleration described in the site's frame. (i.e. you need two gyros in parent and child to get the difference in their readings to get qvel.)

anayebi, 2023-10-30:
Thanks for the speedy reply! Yes, the Mujoco scene initializes correctly without any exceptions. I am getting the SensorReading when observations are collected, after a decision is requested -- though perhaps you mean something different by "update loop"?

Just for my own sanity check, can you describe what you are doing? 
I'll try to replicate it on my end and report back.

Balint-H, 2023-10-30:
I'm assuming you are using ML-Agents. Perhaps as a sanity check, you could create a test MonoBehaviour, that simply prints out the SensorData of the gyro at FixedUpdate (that's what I did). Try it in a scene with only the swimmer in it and no ML-Agents components or other models to begin with.

By the timing of the observation collection I meant if you have taken any specific steps to specify when in the MuJoCo update loop you are collecting observations. However on second thought, this timing concern could only incur a potential frame delay, and would not zero the reading. 
> In case you are interested: By default, I believe you don't have a guarantee that you are collecting observations before or after an update loop with mujoco+ML-Agents. You can specify to take your observations before, after or even within a mujoco update loop at the mujoco control stage by stepping your ML-Agents environment at the appropriate point using MjScene's events, for an example check out the [modular agents environments](https://github.com/Balint-H/modular-agents/blob/main/UnityMjExamples/README.md)).

anayebi, 2023-10-30:
Thanks! I actually see that according to the [documentation](https://mujoco.readthedocs.io/en/stable/XMLreference.html#sensor-gyro), the gyroscope is in fact measuring the angular *velocity* (not acceleration) of the site in local coordinates. I would therefore expect the `SensorReading.y` component of the gyro to line up with the hinge joint's `Velocity` attribute.

I tried your suggestion of attaching a `MonoBehaviour` to the agent to a fresh import and environment without any ML Agents, and I still see it is all 0's in the `SensorReading` of `head/sensors/gyro_0` and `head/sensors/velocimeter_0`, but small negative values in `head/segment_0/joint_0`'s `Velocity` attribute (e.g. `-0.0006122044`).

Here is the code I am attaching to the agent in case, it is useful:
```
using UnityEngine;
using Mujoco;

public class GyroTest : MonoBehaviour
{
    private MjHingeJoint segment0Joint; // joint_0 
    private MjSiteVectorSensor segment0Vel;
    private MjSiteVectorSensor segment0Gyro;
    void Start()
    {
        segment0Joint = GameObject.Find("swimmer3/head/segment_0/joint_0").GetComponent<MjHingeJoint>();
        segment0Vel = GameObject.Find("swimmer3/sensors/velocimeter_0").GetComponent<MjSiteVectorSensor>();
        segment0Gyro = GameObject.Find("swimmer3/sensors/gyro_0").GetComponent<MjSiteVectorSensor>();
    }
    void FixedUpdate()
    {
        // Print the data
        Debug.Log("segment0Vel.SensorReading: " + segment0Vel.SensorReading + " segment0Gyro.SensorReading: " + segment0Gyro.SensorReading + " segment0Joint.Velocity: " + segment0Joint.Velocity);
    }
}
```

anayebi, 2023-10-31:
Thanks for the suggestion! When Ctrl+Shift+Left Click `segment_0` (the first link after the head), I now see non-zero values:
```
segment0Vel.SensorReading: (0.01, 0.00, 0.00) 
segment0Gyro.SensorReading: (0.00, 0.09, 0.00) 
segment0Joint.Velocity: 0.1639257
```

Your helpful description makes sense, and explains to me why I wouldn't expect the gyro to align with the joint velocity. The gyro measures the angular velocity around each of its local X, Y, and Z axes, whereas `qvel` measures the rotation along the given axis, which in this case is the Mujoco Z-axis (Unity Y-axis).

Closing this issue now!

closed 2023-10-31 by anayebi