google-deepmind/mujoco #1146: Missing a .meta file for the mujoco.dll in the unity plugin

ganyuanzhen, 2023-10-30:
I'm using MuJoCo and Unity on Windows 11. However, when I imported the package via git in the Unity editor, it promoted that the mujoco.dll does not have a .meta file and is ignored. 

After I create a .meta file for mujoco.dll manually, the plugin loads successfully. 

So, I think it is better to add .meta files for the runtime libraries.

ganyuanzhen, 2023-10-30:
Or, it is better to create a meta file for the library while copying the library during the installation process.

Balint-H, 2023-10-31:
How you describe it is one way to do it, although the copied files may disappear from the package folder if the project needs to reimport the git package. The way I usually handle this is to copy the mujoco DLL straight to my Project (somewhere in the Assets folder). 

To better support git based installations, maybe the install instructions could be updated with this approach?

Alternatively, maybe including the dll and meta file in the package itself could be appropriate. Or is including a dll like that not good practice?

ganyuanzhen, 2023-11-01:
> How you describe it is one way to do it, although the copied files may disappear from the package folder if the project needs to reimport the git package. The way I usually handle this is to copy [...]

Including a dll file in the repo doesn't seem to be a  very well practise, although it is convenient for users to import. Doing so will lead to the increment of the repo's size at every update, as the dll is a binary file. 

As indicated in the [image], "On Unity version 2020.2 and later, the Package Manager will look for the native library file and copy it to the package directory when the package is imported." But the Package Manager will only copy the library file itself, **not** creating the .meta file at the same time. I think this will be the major problem. 

P.S. I use Unity 2020.3.48f1c1. I'm not sure whether on higher version of Unity this issue will occur or not.

Therefor, the install instructions should be updated to indicate this.

ganyuanzhen, 2023-11-01:
I already open a pr to fix the problem.