google-deepmind/mujoco #1147: Trying to actuate the Manipulator

rishabkr29, 2023-10-30:
[image]
Hi,

I'm a student and I'm trying to use MuJoCo for my project work.

I'm looking for some help with actuating the cable. 

Here is a model which explains my question:

<details>
  <summary>minimal XML</summary>

```XML
<mujoco>
 <compiler autolimits="true"/>
   <worldbody >

      <light diffuse=".5 .5 .5" pos="0 0 10" dir="0 0 -1"/>
      <geom type="plane" size="0.8 0.8 0.1" rgba="1 1 1 1"/>
      
      <body name="Extrusion1" pos="0 0.8 0.01" euler="0 90 0">
          <geom type="box" size="0.02 0.04 0.84" rgba="0.7 0.7 0.7 1"/>
      </body>
      <body name="Extrusion2" pos="0 -0.8 0.01" euler="0 90 0" >
          <geom type="box" size="0.02 0.04 0.84" rgba="0.7 0.7 0.7 1"/>
      </body>
       <body name="Extrusion3" pos="0.8 0 0.01" euler="90 0 0" >
          <geom type="box" size=" 0.04 0.02  0.75" rgba="0.7 0.7 0.7 1"/>
      </body>
      <body name="Extrusion4" pos="-0.8 0 0.01" euler="90 0 0" >
          <geom type="box" size=" 0.04 0.02  0.75" rgba="0.7 0.7 0.7 1"/>
      </body>

      <body name = "1st Link" pos="-0.72 0 -0.05" euler="0 0 0" >
         <joint name="1st Revolute Joint" type = "hinge" axis="0 0 1 " damping=".04"/>
         <geom name="1st Revolute Joint" type="cylinder"  pos="-0.05 0 0.1" euler="0 0 0" size=".04 .01"  rgba="0 .9 0 1"/>
         <geom name="1st Link" type="box" pos="0.15 0 0.1" euler="0 0 0" size=".2 0.04 .01" />   
         <geom name="offset1" type="box" pos="0.15 0 0.1" size=".01 0.09 0.01" /> 
         <site name="l1" pos="0.15 0.09 0.1" rgba="0 .9 0 1"/>   
         <site name="l2" pos="0.15 -0.09 0.1" rgba="0 .9 0 1"/>
      </body>
       
       <body name = "2nd Link" pos="-0.72 0 -0.05" euler="0 0 0">
         <joint name="2nd Revolute Joint" type = "hinge" axis="0 0 1 " damping=".04"/>
         <geom name="2nd Revolute Joint" type="cylinder"  pos="0.35 0 0.1" euler="0 0 0" size=".04 .01"  rgba="0 0 0 1"/>
         <geom name="2nd Link" type="box" pos="0.55 0 0.1" euler="0 0 0" size=".2 0.04 .01"  rgba="0.9 .9 0 1"/> 
         <geom name="offset2" type="box" pos="0.55 0 0.1" size=".01 0.09 0.01" /> 
         <site name="l3" pos="0.55 0.09 0.1" rgba="0 .9 0 1"/>   
         <site name="l4" pos="0.55 -0.09 0.1" rgba="0 .9 0 1"/>  
      </body>

      <body name="Pully1" pos="-0.72 0 -0.05" euler="0 0 0">
         <joint name="Pully1" type="slide"  axis="1 0 0 " damping="1"/>
         <geom name="Pully1" type="cylinder" pos="-0.05 0.2 0.1" euler="0 0 0" size=".01 .01" rgba="0 .9 0 1"/>
         <site name="Pully1" pos="-0.04 0.2 0.1"  rgba="0 0 1 1"/>
         <site name="Motor1" pos="-0.1 0.2 0.1" rgba="0 .9 0 1"/>   
         <geom name="Motor1" type="box" pos="-0.1 0.2 0.1" size=".01 .01 0.01" rgba="0 .9 0.9 1"/>
      </body> 
        <body name="Pully2" pos="-0.72 0 -0.05" euler="0 0 0">
         <joint name="Pully2" type="slide"  axis="1 0 0 " damping="1"/>
         <geom name="Pully2" type="cylinder" pos="-0.05 -0.2 0.1" euler="0 0 0" size=".01 .01" rgba="0 .9 0 1"/>
         <site name="Pully2" pos="-0.04 -0.2 0.1"  rgba="0 0 1 1"/>
         <site name="Motor2" pos="-0.1 -0.2 0.1" rgba="0 .9 0 1"/>   
         <geom name="Motor2" type="box" pos="-0.1 -0.2 0.1" size=".01 .01 0.01" rgba="0 .9 0.9 1"/>
      </body> 
        <body name="Pully3" pos="-0.72 0 -0.05" euler="0 0 0">
         <joint name="Pully3" type="slide"  axis="0 1 0 " damping="1"/>
         <geom name="Pully3" type="cylinder" pos="-0.05 0.4 0.1" euler="0 0 0" size=".01 .01" rgba="0 .9 0 1"/>
         <site name="Pully3" pos="-0.04 0.4 0.1"  rgba="0 0 1 1"/>
         <site name="Motor3" pos="-0.1 0.4 0.1" rgba="0 .9 0 1"/>   
         <geom name="Motor3" type="box" pos="-0.1 0.4 0.1" size=".01 .01 0.01" rgba="0 .9 0.9 1"/>
      </body> 
        <body name="Pully4" pos="-0.72 0 -0.05" euler="0 0 0">
         <joint name="Pully4" type="slide"  axis="0 1 0 " damping="1"/>
         <geom name="Pully4" type="cylinder" pos="-0.05 -0.4 0.1" euler="0 0 0" size=".01 .01" rgba="0 .9 0 1"/>
         <site name="Pully4" pos="-0.04 -0.4 0.1"  rgba="0 0 1 1"/>
         <site name="Motor4" pos="-0.1 -0.4 0.1" rgba="0 .9 0 1"/>   
         <geom name="Motor4" type="box" pos="-0.1 -0.4 0.1" size=".01 .01 0.01" rgba="0 .9 0.9 1"/>
      </body> 
  </worldbody>

  <tendon>
    <spatial name="rope1" range="0 0">
      <site site="l1"/>
      <site site="Pully1"/>
      <site site="Motor1" />
    </spatial>
  </tendon>
   
    <tendon>
    <spatial name="rope2" range="0 0">
      <site site="l2"/>
      <site site="Pully2"/>
      <site site="Motor2" />
    </spatial>
  </tendon>

  <tendon>
    <spatial name="rope3" range="0 0">
      <site site="l3"/>
      <site site="Pully3"/>
      <site site="Motor3" />
    </spatial>
  </tendon>
   
  <tendon>
    <spatial name="rope4" range="0 0">
      <site site="l4"/>
      <site site="Pully4"/>
      <site site="Motor4" />
    </spatial>
  </tendon>

  <actuator>
    <position name="Motor1" joint="Pully1" ctrlrange="0 .1" kp="5000"/>
  </actuator>
   <actuator>
    <position name="Motor2" joint="Pully2" ctrlrange="0 .1" kp="5000"/>
  </actuator>
  <actuator>
    <position name="Motor3" joint="Pully3" ctrlrange="0 .1" kp="5000"/>
  </actuator>
   <actuator>
    <position name="Motor4" joint="Pully4" ctrlrange="0 .2" kp="5000"/>
  </actuator>
  <actuator>
  <position name="1st_link_actuator" joint="1st Revolute Joint" kp="5000" ctrlrange="0 .2" />
</actuator>
</mujoco>

```

</details>

Here is a screenshot / video, illustrating my question:
[image]

Jerold-S, 2023-10-31:
Hey - I am by no means an expert, but I think your tendons need to have proper limits - having the "0 0" just sets them so that that they would behave like infinite springs, if the stiffness was defined. As they are in your model, they are not applying any force to the linkages.

Additionally - and I may be wrong in this - but it looks like you want the two linkages connected together. You either need an equality constraint between them, or more simply, to nest them together in your code so that the joint from the second body references the first body rather than the worldbody. Like this:

<example>

```XML
<mujoco>
  <worldbody >
    <body name = "1st Link" pos="-0.72 0 -0.05" euler="0 0 0" >
      <geom name="1st Link" type="box" pos="0.15 0 0.1" euler="0 0 0" size=".2 0.04 .01" />
      <body name = "2nd Link" pos="-0.72 0 -0.05" euler="0 0 0">
        <joint name="2nd Revolute Joint" type = "hinge" axis="0 0 1 " damping=".04"/>
        <geom name="2nd Link" type="box" pos="0.55 0 0.1" euler="0 0 0" size=".2 0.04 .01"  rgba="0.9 .9 0 1"/> 
      </body>
    </body>
  </worldbody>
</mujoco>
```
</example>

You will also want to consider how the conacts are defined as you have geometry that is overlapping between the two linkages and the surrounding you have created.

yuvaltassa, 2023-11-04:
What @Jerold-S said 🙂

closed 2023-11-04 by yuvaltassa