google-deepmind/mujoco #1151: Getting the GLFW context used by passive viewer to use mjr_readPixels() and record the current viewer window

lok-i, 2023-11-01:
Hi,

I recently migrated my code base to use the mujoco.viewer.launch_passive to visualize my simulation, owing to the custom visualization features. However, since the current glfw context is not exposed, I am unable to grab the frames from this window using mjr_readPixels(), to say record a video. According to #766, this seems impossible and I am aware that an alternative would be to use mujoco.Renderer(). However, I would like to record interactive simulations (using the keyboard callback), hence would like to record the onscreen rendered window from the viewer. Kindly let me know if there is a clean way to do it. Thanks !

lok-i, 2023-11-03:
I  train multiple policies (with a grid of hyperparameters), and for analyzing these policies (typically in the order of 50s), I have test scripts that automatically run them one by one on a test env and save the results (both videos and data logs) for me to view later. Thus, previously, I used this 3rd party wrapper (https://github.com/rohanpsingh/mujoco-python-viewer), where the created glfw context and viewport were exposed, and I could grab the frames and make the video from within python.  Is there a way I can still record my screen from python (not needing human intervention) but without needing the glfw context?

yuvaltassa, 2023-11-04:
If you're recording videos offline, why don't you use regular offscreen rendering like we do in the [tutorial colab](https://github.com/google-deepmind/mujoco#getting-started)?

While I understand what you're asking for, I think separate APIs for interactive and offline rendering make more sense. E.g. what if you're running headless and can't open a glfw window?

lok-i, 2023-11-04:
Yes, I did consider that. Working with the viewer is easier (in the sense you can actively change camera angles and switch visual entities like contact force, collision points, etc., based on what is happening) than the renderer, where I need to set everything apriori to my analysis. Also, I extensively use the `user_scn`  to add custom visualizations (as seen in the video below ), which are available in the viewer. Since `user_scn`  is decoupled from the original scene, I can visualize things without messing up the original scene. Can I do the same in the renderer without book keeping of the custom geoms I want to visualise ? 

https://github.com/google-deepmind/mujoco/assets/54435909/3f21090a-b50b-4958-9144-4e60db398e5f

yuvaltassa, 2023-11-04:
Wow, this kangaroo robot is gorgeous! (I recommend using multiple, directed point lights for better visuals, some of your "steps" are practically invisible with the current lighting)

You can do all of that using offline rendering. In fact, we just added a cell to the end of the tutorial colab, showing how to do [dynamic camera control](https://youtu.be/cE3s_IfO4g4).

Let us know if there is something you want to do but cannot figure out how.

Closing for now.

closed 2023-11-04 by yuvaltassa

lok-i, 2023-11-04:
Thanks for the tip and help. I will try this out. 

The "kangaroo robot" is called Hector (https://youtu.be/NcW-NFwjMh0?si=mC0pTfD7mysvUCaB). We are open-sourcing the model and some controllers from our lab. After a few ongoing design enhancements, I would love to add it to mujoco_menagerie.