google-deepmind/mujoco #1152: Issue importing GMSH file into Flexcomp

auan369, 2023-11-01:
Hi,

I'm a student trying to implement soft-body behaviours on custom shapes in MuJoCo, using the Flexcomp component.

I'm looking for some help with importing gmsh files into the flexcomp component. I repeatedly encounter "_XML Error: Error: All nodes must be in single block_". I have made sure that the file is of GMSH 4.1 version, but apart from that I am unsure of what is causing these issues.

The part that I have issues with:

<details>
  <flexcomp type="gmsh" file = 'C:\mujoco\mujoco-3.0.0\model\plugin\elasticity\asset\cylinder.msh' pos="-9 10 1"
            rgba="0 .7 .7 1" name="softbody1" mass="5">
            <contact condim="3" solref="0.01 1" solimp=".95 .99 .0001" selfcollide="none"/>
            <edge damping="1"/>
            <plugin plugin="mujoco.elasticity.shell">
                <config key="poisson" value="0"/>
                <config key="thickness" value="8e-3"/>
                
                <config key="young" value="3e7"/>
            </plugin>
        </flexcomp>

//full code
```XML
<mujoco model='test'>
    <include file="scene.xml"/>
    <extension>
        <plugin plugin="mujoco.elasticity.solid"/>
    </extension>
    <option solver="CG" tolerance="1e-6" timestep=".001"/>
    <size memory="10M"/>
    <visual>
        <map stiffness="100"/>
    </visual>
    <worldbody>
        <flexcomp type="gmsh" file = 'C:\mujoco\mujoco-3.0.0\model\plugin\elasticity\asset\cylinder.msh' pos="-9 10 1"
            rgba="0 .7 .7 1" name="softbody1" mass="5">
            <contact condim="3" solref="0.01 1" solimp=".95 .99 .0001" selfcollide="none"/>
            <edge damping="1"/>
            <plugin plugin="mujoco.elasticity.shell">
                <config key="poisson" value="0"/>
                <config key="thickness" value="8e-3"/>            
                <config key="young" value="3e7"/>
            </plugin>
        </flexcomp>
        <geom type="mesh" mesh="ball" rgba=".8 .2 .1 1" pos="-2 2 0"/>
    </worldbody>
    <asset>
        <mesh file = "ball10.stl" name="ball" scale="0.2 0.2 0.2"/>
    </asset>
</mujoco>
```

</details>

Here is a screenshot of the error encountered:
[image]

You may find the STL and MSH file in the above code here: https://drive.google.com/drive/folders/1-OpZV_sRu-KlHq89JGPAsPqRqwQr9a9d?usp=sharing

I have added some other mesh files that I have tried but still encounter the same issue.

quagla, 2023-11-01:
Are you able to save the the MSH file with all the nodes in a single block? Currently our parses assumes that, apologies that this was not specified in the documentation.

auan369, 2023-11-01:
Hi @quagla ,

Thank you for your reply! 
I believe that I have saved the MSH file with all the nodes in a single block. I shall list the contents of the ball.msh file below in any case.

```
$MeshFormat
4.1 0 8
$EndMeshFormat
$Entities
0 0 1 2
1 9.067000389099121 -9.875 0 9.932999610900879 -9.125 1 0 0 
1 9.067000389099121 -9.875 0 9.932999610900879 -9.125 1 0 1 1 
2 9.067000389099121 -9.875 0 9.932999610900879 -9.125 1 0 1 1 
$EndEntities
$Nodes
3 16
1 2 0 14
1 9.5 -9.5 0
2
3
4
5
6
7
8
9
10
11
12
13
9.067000389099121 -9.5 0.25
9.282999992370605 -9.125 0.25
9.717000007629395 -9.875 0.75
9.932999610900879 -9.5 0.75
9.5 -9.5 1
9.717000007629395 -9.125 0.75
9.717000007629395 -9.875 0.25
9.932999610900879 -9.5 0.25
9.282999992370605 -9.125 0.75
9.717000007629395 -9.125 0.25
9.067000389099121 -9.5 0.75
9.282999992370605 -9.875 0.25
14 9.282999992370605 -9.875 0.75
1 3 0 2
15 9.355666796366375 -9.25 0.5
16 9.644333203633629 -9.750000000000004 0.5000000000000001
2 3 0 0
$EndNodes
$Elements
2 55
1 2 2 24
1 1 2 3 
2 2 1 13 
3 1 3 11 
4 1 9 8 
5 1 8 13 
6 1 11 9 
7 3 2 12 
8 2 14 12 
9 2 13 14 
10 11 3 10 
11 3 12 10 
12 4 5 6 
13 8 5 4 
14 14 4 6 
15 13 8 4 
16 13 4 14 
17 5 7 6 
18 9 7 5 
19 8 9 5 
20 7 10 6 
21 10 12 6 
22 12 14 6 
23 9 11 7 
24 11 10 7 
1 3 4 31
25 8 5 9 16 
26 5 8 4 16 
27 7 10 11 15 
28 10 3 11 15 
29 2 3 12 15 
30 10 12 3 15 
31 4 8 13 16 
32 14 4 13 16 
33 6 14 12 5 
34 12 6 5 7 
35 8 1 13 9 
36 11 2 1 9 
37 14 6 4 5 
38 6 12 10 7 
39 1 3 2 11 
40 2 13 1 9 
41 12 10 7 15 
42 13 8 9 16 
43 3 2 11 15 
44 4 14 5 16 
45 7 5 12 9 
46 16 5 12 14 
47 16 12 5 9 
48 2 16 12 14 
49 2 12 16 9 
50 13 16 2 14 
51 13 2 16 9 
52 11 15 9 7 
53 11 9 15 2 
54 12 9 15 7 
55 12 15 9 2 
$EndElements
```

quagla, 2023-11-01:
The 3 at the number of the `$Nodes` section means that there are 3 node blocks in this file.

auan369, 2023-11-01:
Thank you for spotting that! May I check how I am able to reduce the number of node blocks to 1?

quagla, 2023-11-14:
This should be possible using Gmsh, which is an open source software. Let me know if that works or feel free to reopen if you have issues.

closed 2023-11-14 by quagla

auan369, 2023-11-21:
I was able to resolve this by manually changing the MSH file after generating it with GMSH. I did it using these 2 steps:
1. Converted all the nodes into 1 node block of 3D nodes 
2. Deleted all elements except for 3D elements so that I only have 1 element block of 3D elements 

I’m not sure if there’s a better way to do this in GMSH, but these steps worked for me so far.

tsreyas, 2023-12-10:
@auan369 I'm having the same issue, albeit with a much bigger mesh. Could you tell me how you would go about grouping the nodes into a single block? I'm not able to find any resources on how to do so.

federico-siemens, 2024-07-24:
Good day. 
i am trying to simulate a soft-body in MuJoCo that i have produced with OpenSCAD. My ultimative goal is to create a more complex structure in OpenSCAD and the simulate it in MuJoCo. For testing purpose i am therefore trying to simulate a simple body from OpenSCAD in MuJoCo so that it exhibits flexibile propertes. First i have created an stl file of the object. Since I am using the flexcomp properties I have tried to transfer the file to a GMSH format using the GSMH software exporting it as an msh file (version 4.1 ASCII). I have created a Volume using Geometry->Elementary entities->Add->Volume. With my python Skript and the corresponding msh file i run into the following error: ValueError: XML Error: Error: All nodes must be in single block. I have tried a lot of different strategies and now I am running out of options and desperatley need help to find a possible solution. Thank you very much! 

import mujoco as mj
import numpy as np
import os
from mujoco.glfw import glfw

#Define Simulation parameters
sim_time = 5 #Simulation time in s
xml_path = "stl_integraton2.xml"#Name of the full path for the XML file
#Define initial Position 
print_camera_config = 1 # set to 1 to print camera config (useful for initializing view of the model)
gmsh_file = "cubic_v4.1.msh"

#MuJoCo model XML
MODEL_XML = f"""
<mujoco>
 
    <asset>
        <mesh name="Box" file="{gmsh_file}" />
    </asset>

    <option timestep="0.01"/>
    <extension>
        <plugin plugin="mujoco.elasticity.solid" />
    </extension>
    <size memory="100M"/>
    
    <worldbody>
        <geom name="floor1" type="plane" pos="0 0 0.3" size="100 100 0.5" euler="0 0 0" rgba="0.9 0.9 0.9 1"/>
        <light diffuse="1 1 1" pos="0 0 100" dir="0 0 -1"/>
        
         
        <flexcomp name="stl_file" type="gmsh" file="{gmsh_file}" dim="3" pos="0 0 10" rgba=".3 .2 .2 1" mass="100">
            <edge equality="true" damping="0.3"/>
            <contact condim="3" solref="0.03 1" solimp=".95 .99 .0001" selfcollide="none"/>
            
            <plugin plugin="mujoco.elasticity.solid">
               <config key="poisson" value="0.2"/> 
               <config key="young" value="1e5"/>
            </plugin>
            
        </flexcomp>

    </worldbody>
</mujoco>
    """

# Save the XML to a file
with open(xml_path, "w") as f:
    f.write(MODEL_XML)
#...

msh file: 
$MeshFormat
4.1 0 8
$EndMeshFormat
$Entities
0 0 1 2
1 0 0 0 5 5 5 0 0 
1 0 0 0 5 5 5 0 1 1 
2 0 0 0 5 5 5 0 1 1 
$EndEntities
$Nodes
3 8 1 8
2 1 1 8
1
2
3
4
5
6
7
8
0 5 5 0 0
5 0 5 0 0
5 5 5 0 0
0 0 5 0 0
0 0 0 0 0
5 5 0 0 0
5 0 0 0 0
0 5 0 0 0
3 1 0 0
3 2 0 0
$EndNodes
$Elements
2 18 1 18
2 1 2 12
1 1 2 3 
2 2 1 4 
3 6 1 3 
4 1 5 4 
5 5 1 8 
6 1 6 8 
7 2 6 3 
8 5 2 4 
9 2 5 7 
10 6 2 7 
11 5 6 7 
12 6 5 8 
3 1 4 6
13 6 2 5 7 
14 6 1 5 3 
15 5 1 4 2 
16 2 6 5 3 
17 1 6 5 8 
18 5 3 1 2 
$EndElements

auan369, 2024-08-04:
For your case the following MSH file might work:

$MeshFormat
4.1 0 8
$EndMeshFormat
$Entities
0 0 1 2
1 0 0 0 5 5 5 0 0
1 0 0 0 5 5 5 0 1 1
2 0 0 0 5 5 5 0 1 1
$EndEntities
$Nodes
1 8 1 8
3 1 1 8
1
2
3
4
5
6
7
8
0 5 5 0 0
5 0 5 0 0
5 5 5 0 0
0 0 5 0 0
0 0 0 0 0
5 5 0 0 0
5 0 0 0 0
0 5 0 0 0
$EndNodes
$Elements
1 18 1 18
3 1 4 18
1 1 2 3
2 2 1 4
3 6 1 3
4 1 5 4
5 5 1 8
6 1 6 8
7 2 6 3
8 5 2 4
9 2 5 7
10 6 2 7
11 5 6 7
12 6 5 8
13 6 2 5 7
14 6 1 5 3
15 5 1 4 2
16 2 6 5 3
17 1 6 5 8
18 5 3 1 2
$EndElements

auan369, 2024-08-04:
@federico-siemens

auan369, 2024-08-04:
> @auan369 I'm having the same issue, albeit with a much bigger mesh. Could you tell me how you would go about grouping the nodes into a single block? I'm not able to find any resources on how to do so.

@tsreyas sorry for the much delayed response. Please let me know if you still need help on this

mxhrnxz, 2025-05-26:
> [@tsreyas](https://github.com/tsreyas) sorry for the much delayed response. Please let me know if you still need help on this

Hello!
I'm facing the same problem. Would you please share your solution with us? I really appreciate it.