google-deepmind/mujoco #1159: Sync()  runs slowly in Mujoco 3.0.0 python bindings

hxrkpib, 2023-11-02:
Hi,

When I run a minimal example in mujoco 3.0.0 with 1 ms timestep, I found the sync() takes 6 ms. Before that, load the same model and use same timestep(1 ms) in mujoco 2.3.3(of course there is no sync() yet), everything is alright.

```
with mujoco.viewer.launch_passive(m, d) as viewer:
        # Close the viewer automatically after 30 wall-seconds.

        while viewer.is_running():
            step_start = time.time()

            # mj_step can be replaced with code that also evaluates
            # a policy and applies a control signal before stepping the physics.

            mujoco.mj_step(m, d)

            # Example modification of a viewer option: toggle contact points every two seconds.
            with viewer.lock():
                viewer.opt.flags[mujoco.mjtVisFlag.mjVIS_CONTACTPOINT] = int(
                    d.time % 2)

            # Pick up changes to the physics state, apply perturbations, update options from GUI.
            start_time = time.time()
            viewer.sync()
            print("sync() duration:",time.time()-start_time,"s")
            # Rudimentary time keeping, will drift relative to wall clock.
            time_until_next_step = m.opt.timestep - (time.time() - step_start)
```

[image]

kevinzakka, 2023-12-03:
Cannot reproduce this on my end. Can you provide more details about your system (OS, graphics card, python version) and also attach a model?

kevinzakka, 2024-01-03:
Hi @hxrkpib, any updates?

kevinzakka, 2024-08-30:
Closing because stale.

closed 2024-08-30 by kevinzakka