google-deepmind/mujoco #1160: Create custom XML shortcuts

mavex857, 2023-11-02:
Hi,

I'm a and I'm trying to use MuJoCo for simulating a muscle prototype that is being developed with my university at the moment.I would like to know if it is possible to create custom XML shortcuts, for example for this specific actuator.
So for example if I have to use this a whole lot of times:
```XML
...
<actuator>
  <general><name="act_0" tendon="tnd_left" gaintype="user" acttype="user" lengthrange="0.2 1.7"... ></general>
</actuator>
...
```
Can I create something like this:
```XML
...
<actuator>
  <my_actuator><name="act_0" tendon="tnd_left"></my_actuator>
</actuator>
...
```

I couldn't find this in the docs and would be thankful for any suggestions :)
Cheers

yuvaltassa, 2023-11-02:
https://mujoco.readthedocs.io/en/latest/modeling.html#default-settings

closed 2023-11-02 by yuvaltassa

mavex857, 2023-11-06:
Thanks for pointing me to the docs, sorry that I missed that before asking here!
However I cannot get it to work correctly, the following snippet of my XML file won't compile:
```    
<default class="main">
        <actuator>
            <general gaintype="user" user="1 1" lengthrange="0.3 1"></general>
        </actuator>
</default>
```
It throws 
```
ValueError: XML Error: Schema violation: unrecognized element

Element 'actuator', line 4
```

yuvaltassa, 2023-11-06:
You're doing it wrong. Read some models from the [MuJoCo Menagerie](https://github.com/google-deepmind/mujoco_menagerie) (e.g., in the videos in [this](https://mujoco.readthedocs.io/en/latest/models.html) page) for good examples of using defaults. Then read the basic [documentation](https://mujoco.readthedocs.io/en/latest/computation/index.html#actuation-model) again. Once you understand the mechanism, you might want to read [this section](https://mujoco.readthedocs.io/en/latest/modeling.html#shortcuts), which explains some specific gotchas with respect to actuator defaults.