google-deepmind/mujoco #1161: Feature comparison between Omnivers's IsaacSim and gpu-accelerated Mujoco 3.0.0

AndrePatri, 2023-11-02:
Hi everybody,

I'm a PhD student in robotics and I'm currently using Omniverse IsaacSim for training some RL policies for locomotions.

I would really love to try MuJoCo, however I was wondering how it compares wrt IsaacSim not in terms of performance (this is already covered by your release notes), but in terms of API completeness.
Specifically, 
- Is MuJoCo already equipped with something similar to Isaac's cloner/grid clone API to achieve environment duplication? 
- Does MuJoCo have API to filter collisions between these environments? 
- Can I import heightmaps in MujoCo to simulate complex environments?
- How much of MuJoCo 3.0 is on GPU? Get I directly get all quantities on GPU (robot position, etc..) similarly to what is possible on IsaacSim by enabling the so-called gpu_pipeline?

This is sort of a generic question, but I think that users coming from IsaacSim/IsaacGym worlds might find the answer particularly useful.

erikfrey, 2023-11-03:
Hi Andrea,

Good questions!  I will answer each one in turn:

> Is MuJoCo already equipped with something similar to Isaac's cloner/grid clone API to achieve environment duplication?

MuJoCo - via MJX - plays well with all the primitives that [JAX](https://github.com/google/jax) offers - `scan`, `vmap`, `grad` and so on.  These primitives are quite powerful and expressive - please see the [Domain Randomization section](https://colab.sandbox.google.com/github/google-deepmind/mujoco/blob/main/mjx/tutorial.ipynb#scrollTo=65mIPj6DQNNa) of our tutorial for an example of cloning environments with random physics properties in each environment, using just a few lines of code.

Note that this is a bit different than Isaac's cloning as I understand it from the documentation - in MJX we create N truly distinct environments (independent universes) rather than spatially organizing N objects in the same universe. 

> Does MuJoCo have API to filter collisions between these environments?

Yes - collisions are isolated to individual environments, since each environment is truly independent.  Beyond that, `MJX` supports filtering collisions using `contype`, `conaffinity` and `pair` elements.  See MuJoCo's documentation on [computing collisions](https://mujoco.readthedocs.io/en/latest/computation/index.html#collision) for an in-depth explanation of filtering within environments.

> Can I import heightmaps in MujoCo to simulate complex environments?

Heightmaps are coming!  For now, I'd recommend meshes or building them up via other primitives.

> How much of MuJoCo 3.0 is on GPU? Get I directly get all quantities on GPU (robot position, etc..) similarly to what is possible on IsaacSim by enabling the so-called gpu_pipeline?

It's 100% accelerated.  MJX is specifically designed to work in tandem with learning algorithms that also live on device.

Stepping back to answer your question more broadly:

* At the time of writing, IsaacSim has a more extensive feature set, in particular better support for large scale scenes.  If you want to simulate automation in a warehouse, for example, I'd recommend IsaacSim!
* Philosophically, I think it's fair to think of MJX more as a library, and IsaacSim more a component of a platform offering.  Concretely this means:
  * MJX is fully open source, available to use in your project via [Apache 2.0 License](https://github.com/google-deepmind/mujoco/blob/main/LICENSE).  We welcome hacking on MJX, accept PRs from the community, and so forth.
  * MJX strives to be vendor-agnostic: we test and benchmark on AMD, TPU, even Apple hardware (although that's still a work in progress!).
  * We've designed MJX to be embeddable into your project, not the other way around!
  * But as a result we have fewer batteries included - e.g. we don't have powerful IDE integrations as Omniverse, and so on.

I think MJX's sweet spot right now is RL/robotics research in relatively simple (e.g. room-scale) scenes, where you ultimately want to transfer your policy onto a real robot.

AndrePatri, 2023-11-03:
Thanks @erikfrey, @saran-t for the answers. That's great news for me!

AndrePatri, 2023-11-03:
@saran-t  does this mean that when getting observations from the environment (i.e. the simulation) I will have to manually copy them on GPU for my agent? 
Also, normally IsaacSim data is organized in tensors which are n_envs x data size, which is useful when working with parallel simulation. Do I need to take care of this myself by merging data from each independent simulation, or  is the data I get already formatted in some way?

saran-t, 2023-11-03:
If you don't use `mjx.device_get_into` then nothing's copied to CPU.

It's worth reiterating that MuJoCo and MJX are independent implementations of the same physical model. The two share the same model compiler as an entry point, which converts MJCF into `mjModel`. You can then choose to either timestep this using classic MuJoCo, or use `mjx.device_put` to place it on the GPU and use it as part of a bigger training loop, say.

AndrePatri, 2023-11-03:
I see, @saran-t, this addresses my concerns.  Thanks a lot for the detailed replies. I think I'll close this for now.

closed 2023-11-03 by AndrePatri

lok-i, 2023-11-18:
Hi @erikfrey and @saran-t, any updates on the height field geom integration for MJX ? Any rough dates when one can be expecting it ? In my projects I set my height fields dynamically in code every env reset, will this be possible to do with the upcoming hfield feature integration ?