google-deepmind/mujoco #1168: Key-frames

TianhongZhou, 2023-11-05:
Hi,

I'm a student  and I'm trying to use MuJoCo for project.

I'm looking for some help with Unity plugin.

I have installed the MuJoCo plugin in Unity and have successfully import the model from MuJoCo xml. This is what I get so far:
[image]

I need to add the key-frames to it. I have 2 txt files, the first one is a list of time stamps, and the second one is the data of rotations of the joint for each key-frame. But I do not know how can I import those into the Unity in order to have the same animation that I had in MuJoCo. Is there any suggestions?

Balint-H, 2023-11-06:
If you have the full qpos and qvel vectors of your keyframes, you can assign them to the MjData.

Quick example code (you have to call it after the MjScene has been initialized, a good way to do it is to use the MjScene callback functions).

```c#

public void SetMjKeyFrame(double[] keyFrameQPos, double[] keyFrameQVel)
{
    for(int i=0; i<keyFrameQPos; i++)
    {
        MjScene.Instance.Data->qpos[i] = keyFrameQVel[i];
    }

    for(int i=0; i<keyFrameQvel; i++)
    {
        MjScene.Instance.Data->qvel[i] = keyFrameQVel[i];
    }
}

public void TrackReferenceData()
{
    var curQPos = GetQPos(Time.fixedTimeAsDouble) ; // Function you need to define that grabs the best QPos vector from the reference data
    var curQVel = GetQVel(Time.fixedTimeAsDouble)  // Function you need to define that grabs the best QPos vector from the reference data
    SetMjKeyFrame(curQPos, curQVel);
}
```

You may load your keyframes with the text file parser of your choice (storing it in JSON might be swiftest). Create a MonoBehaviour that loads your keyframes and stores them in a ragged array, then subscribes the `TrackReferenceData()` function to `MjScene.Instance.ctrlCallback`.

Let me know if you have trouble with that, or if you have a huge amount of data that is not great to be stored with JSON.

TianhongZhou, 2023-11-06:
@

TianhongZhou, 2023-11-06:
I have several questions.

The first one is that the appoarch that I tried yesterday is to write a function that to assign the angle of each joint to the "Control" in MjActuator for each actuators at each key-frames. After I doing this, the humanoid can somehow moved (but with completely wrong motion), I am wondering is this also an appoarch to let it move? Or I should do something like you mentioned above?

The second one is that here are two errors that I get when I run the scene:
[image]
[image]
I am wondering is it caused due to I have not pass a MjData to it? If so, how can I do it to deal with this error and what should the MjData be? Should it be like "the full qpos and qvel vectors of keyframes" that you mentioned above?

Thank you so much for your help in advance!

Balint-H, 2023-11-06:
Actuators are used for a different goal than keyframes. Actuators apply forces to the degree of freedom they are paired with, which results in continuous motion. Keyframes can be used to discontinuously jump to a given state. What kind of behaviour are you looking to reproduce/simulate?

As far as the error messages go: are they alone or accompanied by another error? You dont need to pass MjData to the creation of the components, that should be handled automatically. If another error prevented the setting up of the MjScene that could result in this error too.

Can you run a simple mujoco scenes (e.g. just a mujoco cube) with no issue?

TianhongZhou, 2023-11-06:
I am trying to simulate the motion similar to the one in this project:
https://xbpeng.github.io/projects/InterPhys/index.html

The motion I have looks quite good in MuJoCo. However, I want to render it more fancy (like the one shown in that project video) and therefore I am trying to do so using Unity plug-in. But the data that I have is only a list of angles to each joint for each keyframe (from MuJoCo). Do I have to get the data of "the full qpos and qvel vectors" that you mentioned in order to do so?

For the error messages, that two messages comes together, and no any other error occurs, like this:
[image]

Balint-H, 2023-11-06:
Yeah if you just want to play back a prerecorded animation then you need to set the qpos on every frame, you dont need to use actuators. Are those errors from a minimal scene? Just a mujoco cube, and an MjGlobalSettings.

TianhongZhou, 2023-11-07:
No I haven't try that yet. If I wanna build a minimal scene, do I need to build it in mujoco first and then import it to unity? Or can I directly create it in unity?

Balint-H, 2023-11-07:
Both can work, but perhaps the easiest way to ensure you have all components is to write a minimal xml and import it, then press play. If that doesnt run it might mean you have an issue with the mujoco.dll. It should be placed either in the Assets folder or the MuJoCo package folder, and should match the version of the package.

TianhongZhou, 2023-11-07:
This is the data loader that I have so far:
`public class SetControl : MonoBehaviour
{
    public TextAsset qpos;
    public TextAsset qvel;
    public double[] keyFrameqpos;
    public double[] keyFrameqvel;

    void Start()
    {
        string qposText = qpos.text;
        qposText = qposText.Replace("\n", "").Replace(" ", "");
        string[] qposNumbers = qposText.Split(new char[] { ',' }, StringSplitOptions.RemoveEmptyEntries);
        keyFrameqpos = qposNumbers.Select(double.Parse).ToArray();

        string qvelText = qvel.text;
        qvelText = qvelText.Replace("\n", "").Replace(" ", "");
        string[] qvelNumbers = qvelText.Split(new char[] { ',' }, StringSplitOptions.RemoveEmptyEntries);
        keyFrameqvel = qvelNumbers.Select(double.Parse).ToArray();
    }

    public unsafe void SetMjKeyFrame(double[] keyFrameQPos, double[] keyFrameQVel)
    {
        for (int i = 0; i < keyFrameQPos.Length; i++)
        {
            MjScene.Instance.Data->qpos[i] = keyFrameQVel[i];
        }

        for (int i = 0; i < keyFrameQVel.Length; i++)
        {
            MjScene.Instance.Data->qvel[i] = keyFrameQVel[i];
        }
    }

    public void TrackReferenceData()
    {
        SetMjKeyFrame(keyFrameqpos, keyFrameqvel);
    }
}`

I am wondering where and how can I call the TrackReferenceData(). 
[image]

I tried to call this like this but it failed, also I try to use ctrlCallback that you mentioned above, but I do not how to use ctrlCallback to call it. The only place that I found the ctrlCallback is here:
[image]

Balint-H, 2023-11-07:
You can perform the subscription in your SetControl component's Start() function.
```c#
public void Start()
{
    //... Whatever came before

    // Subscribe our handler to the control callback
    MjScene.Instance.ctrlCallback += (_, _) => TrackReferenceData()
}
```
Also, I realized you might need to call `MujocoLib.mj_forward(MjScene.Instance.Model, MjScene.Instance.Data)` to make sure other parts of the data structure are updated to reflect changes to qpos and qvel. I added this change to the original comment.

 You can remove the edits you made to MjScene. The callback, and all handlers that have been will be invoked at the location you identified, at each timestep.

Multiple handlers can be subscribed to the callback, thats why use the += operator; we add a new handler to the list of handlers executed when the callback is called. The callback supplies additional arguements to its handlers we don't need. Thats why I wrapped our method with `(_, _) => TrackReferenceData()` , which discards the two arguements.

I noticed you only load a single keyframe. Setting that on every timestep will just give you a static pose. If you just want to jump to that pose and let the simulation run, you don't need to subscribe to the control callback at all, you can just call your method at the frame you want the reset. If its on the very first frame, to guarantee that the MuJoCo scene is ready, you can subscribe the tracking method to `MjScene.Instance.postInitEvent`, which is called only once after the scene is initialized.

If instead you want to play back a series of keyframes (essentially an animation) you need an array of arrays of your qpos (and potentially qvel), and to use the ctrlCallback.

As far as the exception goes, if you still get it at start up you might want to downgrade to 2.3.7 for now, as I confirmed I get the same issue on 3.0.0. In anticipation to an issue you might run into with 2.3.7, you need to set the "Size" property of the MjGlobalSettings in the scene to -1.

Unsafe code does indeed need to be enabled, there is a checkbox in the player settings called "Allow unsafe code". You might need to expand some of the dropdowns.

TianhongZhou, 2023-11-07:
Thank you so much! I have changed the code to have double[][] and handle the animation. This is what I get so far:

https://github.com/google-deepmind/mujoco/assets/93419666/dcd4833b-8175-4d2d-9d4c-5faf7ccf471d

which is much closer to the animation that I had in MuJoCo. But still have a few problems:

Part of the humanoid went to underground, which is not expected. Is it something wrong with my data or with my code or something else? Also, it's hand is kind of floating on the ground, but the animation that I had in MuJoCo is fine, why is it?

Also, I am still getting these two error messages:
[image]

I tried to do what you suggested before -- to load a simpler scene instead. I tried to loaded the balloons.xml that is provided in your repo of xml but I still get some other different errors:
[image]
I did copy the dll into the package folder like this:
[image]

Balint-H, 2023-11-08:
Could you try downgrading to mujoco 2.3.7? Along with the 2.3.7 dll. Or are these results with that already? The way you added the dll to the package looks correct.

TianhongZhou, 2023-11-08:
Yes, it works! No error messages any more and the animation looks quite good! Thank you soooooo much for your help!

I have another quite quick question. I am wondering is there any way to speed up the animation without changing the qpos and qvel? Or I can only do it by changing the data that I have?

Balint-H, 2023-11-08:
The speed of the animation will depend on the ratio of Unity timestep (you can set this) and the original MuJoCo timestep you recorded. You can also apply a transformation on how you look up the data in your arrays (e.g. you can skip every other frame for twice as fast)

Glad it works!

closed 2023-11-22 by kevinzakka

TianhongZhou, 2024-01-19:
Hi, I am wondering if is there any way to have two humanoids with different motion data in the same scene at the same time?