google-deepmind/mujoco #1169: Can I speed up the integration?

nor-code, 2023-11-05:
Hi, @yuvaltassa
I try to train the robot to move along a given trajectory. I use Python, in the code below I create the environment in which the simulation will take place, can I speed up the integration time (x1.5 x3)? There are exists option, that accelerate integration?
(i.e. <option accelerateTime = "1.5"/> ) Because in simple Mujoco(in sense without python library) i can accelerate simulation.

[image]

yuvaltassa, 2023-11-05:
Try to increase the timestep.
Why are you using RK4?
Read this

https://mujoco.readthedocs.io/en/latest/computation/index.html#numerical-integration

closed 2023-11-05 by yuvaltassa

nor-code, 2023-11-05:
@yuvaltassa , sorry, i think You don't understand what i mean exactly.  there is ability run simulation faster than real time mode in 2 or more times (as rewind video)? I use RK4 because need high precition.

yuvaltassa, 2023-11-05:
You are right, I do not understand what you are asking. 
Your latests clarification did not help.
Feel free to try asking again.

nor-code, 2023-11-05:
@yuvaltassa, let considered simple diff. eq. x'' = 0 (stone fall with 0 initial velocity). Suppose that has falling 5 second and we want know it velocity/coordinate every 0.005 [s]. For simulation of this process then we don't wait all 5 second. our CPU can solve it faster. And as I understand it, the engine solves equations in approximately real time, i.e. 1 second in the simulation = 1 second in our world? if this is so, then why is there no way to speed up the numerical integration. This will reduce training time.
P.S. I don’t know English well, so please excuse me if I express myself incorrectly.

yuvaltassa, 2023-11-06:
Simulation in MuJoCo is usually much faster than real time (for example a humanoid simlates at ~200x real-time on a single CPU thread). Where are you seeing anything that limits it to real time? In the `simulate` viewer there is time keeping which introduces pauses in order to try and match real-time (or slower), but there is no such pausing anywhere else.

nor-code, 2023-11-06:
@yuvaltassa thanks, now it is clear!