google-deepmind/mujoco #1170: Set initial joint positions for a RL episode

GianB212, 2023-11-05:
Hello! 

I'm working on a RL project that involves a 6-joint manipulator robot. At the start of each training episode, I would like to randomize the starting angle of each joint, so in short I believe I need to change the state of the model to specific values.

The problem is that modifying the respective `data.qpos` doesn't work, as the change seems to vanish as soon as the simulation starts. I've also tried modifying the `model.body_quat` variable, which worked in order to rotate each link, but unfortunately it also rotates the joint limits.

Is there another way to set initial conditions for the joints with actuators? Thank you.

My model is the following:

`<mujoco model="kuka_kr16_2">
    <asset>
        <mesh name="base_link" file="base_link.stl" />
        <mesh name="link_1" file="link_1.stl" />
        <mesh name="link_2" file="link_2.stl" />
        <mesh name="link_3" file="link_3.stl" />
        <mesh name="link_4" file="link_4.stl" />
        <mesh name="link_5" file="link_5.stl" />
        <mesh name="link_6" file="link_6.stl" />
    </asset>
    <worldbody>
        <geom type="mesh" mesh="base_link" rgba="0.1 0.1 0.1 1" />
        <body name="link_1" pos="0 0 0.675" >
            <inertial pos="1 0 0" mass="2" diaginertia="0.01 0.01 0.01" />
            <geom type="mesh" mesh="link_1" rgba="0.95 0.36 0.1 1"/>
            <joint name="joint_a1" pos="0 0 0" axis="0 0 -1" limited="true" range="-3.22886 3.22886" type="hinge"/>
            <body name="link_2" pos="0.26 0 0">
                <inertial pos="0 0 0" mass="2" diaginertia="0.01 0.01 0.01" />
                <joint name="joint_a2" pos="0 0 0" axis="0 1 0" limited="true" range="-2.70526 0.610865" />
                <geom type="mesh" mesh="link_2" rgba="0.95 0.36 0.1 1"/>
                <body name="link_3" pos="0.68 0 0">
                    <inertial pos="0 0 0" mass="2" diaginertia="0.01 0.01 0.01" />
                    <joint name="joint_a3" pos="0 0 0" axis="0 1 0" limited="true" range="-2.26893 2.68781" />
                    <geom type="mesh" mesh="link_3" rgba="0.95 0.36 0.1 1"/>
                    <body name="link_4" pos="0.67 0 -0.035">
                        <inertial pos="0 0 0" mass="2" diaginertia="0.01 0.01 0.01" />
                        <joint name="joint_a4" pos="0 0 0" axis="-1 0 0" limited="true" range="-6.10865 6.10865" />
                        <geom type="mesh" mesh="link_4" rgba="0.95 0.36 0.1 1" />
                        <body name="link_5" pos="0 0 0">
                            <inertial pos="0 0 0" mass="2" diaginertia="0.01 0.01 0.01" />
                            <joint name="joint_a5" pos="0 0 0" axis="0 1 0" limited="true" range="-2.26893 2.26893" />
                            <geom type="mesh" mesh="link_5" rgba="0.95 0.36 0.1 1"/>
                            <body name="EE" pos="0 0 0">
                                <inertial pos="0 0 0" mass="2" diaginertia="0.01 0.01 0.01" />
                                <joint name="joint_a6" pos="0 0 0" axis="-1 0 0" limited="true" range="-6.10865 6.10865" />
                                <geom type="mesh" mesh="link_6" rgba="0.95 0.36 0.1 1"/>
                            </body>
                        </body>
                    </body>
                </body>
            </body>
        </body>
    </worldbody>
    <actuator >
                <position name="a1_pos" joint="joint_a1" ctrllimited="true" ctrlrange="-300.22886 3.22886" gear="1" kp="500"/>
                <position name="a2_pos" joint="joint_a2"/>
                <position name="a3_pos" joint="joint_a3"/>
                <position name="a4_pos" joint="joint_a4"/>
                <position name="a5_pos" joint="joint_a5"/>
                <position name="a6_pos" joint="joint_a6"/>

    </actuator>
</mujoco>`

yuvaltassa, 2023-11-06:
Setting `data.qpos` is correct and the values do not "vanish". You probably have an `mj_resetData` call after you set the joints.

closed 2023-11-06 by yuvaltassa

hooyahliu, 2024-02-26:
> Setting `data.qpos` is correct and the values do not "vanish". You probably have an `mj_resetData` call after you set the joints.

Hi!
I encountered the same problem.
My test program is as follows:
```
for name, value in qpos_new.items():

    joint_id = joint_name2id[name]
    data.qpos[joint_id] = value
mujoco.mj_resetData(model, data)
print('qpos: ', np.round(data.qpos, 2))
```

The initial qpos was not set to the value I expected, but changed back to 0.
Did I use it wrong?

Thanks in advance!