google-deepmind/mujoco #1171: ArgumentNullException and Failed to create Mujoco runtime

Wafeechan, 2023-11-06:
Hello, I had the same problem as issue #315 when trying out MuJoCo. In this case, I was trying to do the ball on the tilted plane stuff. I am currently testing it out on unity 2022.3.9f1 personal on Windows 10, using MuJoCo 3.0.1. The error was:
**"ArgumentNullException: Value cannot be null. Parameter name: native"**

Then followed by a spam of:
 **"NullReferenceException: Failed to create Mujoco runtime. Mujoco.MjScene.StepScene () (at D:/..../mujoco/unity/Runtime/Components/MjScene.cs:345)"**

I have cloned a repository of the plugin from here: https://github.com/google-deepmind/mujoco/tree/main, and manually imported the package from disk. Do I need the package to be on the same disk partitions? (As in I downloaded the package on D: and imported it from there, while my user directory was on C:).

Secondly, it is probably because I didn't understand some terms, but I did "Expand the zip archive to a directory called MuJoCo in your user directory, and copy the file MuJoCo\bin\mujoco.dll." By this does it mean that I should make a new user directory called "MuJoCo", or should it be automatically created like when I import the package? Because I manually created an empty MuJoCo folder directory for it.

Thirdly, what does it actually mean by "The library can also be copied into any location under your project’s Assets directory."? Does it mean that I have to manually copy paste the mujoco.dll file to the project's Assets directory? Or was there actually another way to do this?

Also lastly, the line "On Unity version 2020.2 and later, the Package Manager will look for the native library file and copy it to the package directory when the package is imported.". I was actually assuming this do be done, but after importing the plugin, a warning prompt appeared telling me there's no bin directory on the user directory, which is why I did it manually as implied on the second and third question.

Sorry for the lengthy question, but I hope my question can be solved so I can finally start exploring MuJoCo for my final year project. It may be something dumb on my part, but any kind of help is appreciated. Thank you.

Balint-H, 2023-11-06:
> Secondly, it is probably because I didn't understand some terms, but I did "Expand the zip archive to a directory called MuJoCo in your user directory, and copy the file MuJoCo\bin\mujoco.dll." By [...]

Yup this one is done manually.

> Thirdly, what does it actually mean by "The library can also be copied into any location under your project’s Assets directory."? Does it mean that I have to manually copy paste the mujoco.dll file to the project's Assets directory? Or was there actually another way to do this?

Its optional. Your project should have one reference to the mujoco.dll library, either placed in the package folder, or in your Assets folder (or subfolders).

> Also lastly, the line "On Unity version 2020.2 and later, the Package Manager will look for the native library file and copy it to the package directory when the package is imported.". I was [...]

If you have manually copied the dll, it should be fine. I have yet to try Unity plugin with MuJoCo 3, but sometimes there is a bit of a lag before all issues with a release are ironed out on the Unity plugin side. You may want to downgrade to 2.3.7 (which I can verify works, if you downgrade, make sure the package and dll versions match). I'll eventually try to update to MuJoCo 3 in the near future, you could message in a week to check in? 

Here's an example Unity project that has worked as fresh installs on multiple computers for me:
https://github.com/Balint-H/modular-agents/tree/main/UnityMjExamples

Wafeechan, 2023-11-10:
Sorry for the late reply, but thank you for the feedback. I will try to do it as soon as possible.

vidurvij-apptronik, 2023-11-30:
I am facing the same issue with MuJoCo. It seems like the plugin is not working with version 3.0.1. I have to downgrade to 3.0.1 to 2.3.7 to run the simulation.

Balint-H, 2023-12-04:
Hello, I can confirm there were a couple issues with the 3.0.1 release of the Unity plugin. You can stick with 2.3.7 for the time being until the next full release, alternatively the plugin from the main branch seems to be working on all my scenes.

vidurvij-apptronik, 2023-12-15:
Hi, So i just tried this again with the new MuJoCo 3.1.0 plugin with the Unity Package version 3.1.0. And its giving me the same error. 2.3.7 is still stable. I need the new version because I want to integrate into our CI/CD pipeline and it seems when I do the following [modification](https://github.com/google-deepmind/mujoco/pull/1254) in 2.3.7, Unity crashes but it seems to work with 3.x.x+.

Balint-H, 2023-12-19:
Hi, thanks for the patience. I tried both 3.1.0 and 3.1.1 today and they both compile and run for the basic humanoid scene. Could you give reimporting the package a go? Also, just in case, but make sure to upgrade the MuJoCo binary along the plugin too.

Balint-H, 2024-01-03:
Please let us know if the issue persisted for you with 3.1.1+, otherwise it can be marked as closed.

closed 2024-01-03 by kevinzakka

vidurvij-apptronik, 2024-01-12:
@Balint-H @kevinzakka  This issue still persists!

Balint-H, 2024-01-12:
Hello, sorry to hear that. In this case can you confirm that the same issue happens after importing an MJCF from the example models and running without any custom scripts? (e.g. humanoid.xml)

vidurvij-apptronik, 2024-01-16:
Yes! Without any custom scripts and no edits to the mujoco package, the issue still presistes. I am using Ubuntu 20 with Unity 2022 LTS latest version.

Balint-H, 2024-01-16:
I see, if I remember, you mentioned using the mujoco dll; could you make sure you have the linux library as well? It will be called `libmujoco.so`. You can get the linux release of MuJoCo for it specifically. You might also need to rename this binary file so its extension is correct.

vidurvij-apptronik, 2024-01-16:
Yep I can confirm that I downloaded the .so file. I am using the method of saving the binaries in the location in the linux machine where it automatically detects the .so file.

AndreTInfante, 2024-03-01:
@Balint-H Just to confirm, I am having the same problem as described, including with the example project you linked, on both the current and older (2.3.6) version of mujoco, on windows 10 on Unity 2022.

vidurvij-Unity, 2024-03-01:
@Balint-H Any progress on this?

Balint-H, 2024-03-01:
Apologies for the delay, I only just set up with a Linux Unity Editor (previously I only ran built scenes on Linux).

On version 3.1.2, scenes run for me on Ubuntu and Unity 2023.2.11 without exceptions. It works with the .so file either in the Asset folder or the package itself. Could you please confirm from a file explorer (not Unity's) what is the extension of the libmujoco library file in your project? It should be only ".so".

If you copied libmujoco manually from the release, make sure to use the version called libmujoco.so.3.x.x, and then rename it to just libmujoco.so, once its imported to your project.

[image]
e.g. copy the bottom one to your project, then rename it to match the top one. The BinaryRetriever should do this automatically if it can find the .so files, but I can confirm it works manually too.

Let me know if this still does not work! If you can share a unity project repo where the issue is reproduced, I can take a look and see if there's anything that stands out.