google-deepmind/mujoco #1172: Need help in simulating a robot with closed kinematic loop in MuJoCo

aamodbk, 2023-11-07:
Hi,

I'm a student and I'm trying to use MuJoCo for simulating a robot with its SDF exported from Fusion 360. I have used a tool called sdf2mjcf to convert SDF to MuJoCo's XML format. Inorder to successfully export the model to mJCF using the library I needed to remove a joint which formed the closed loop. Hence I removed one of the revolute joints in the loop for exporting it to MJCF and then tried to add the joint by writing an equality constraint between the two links.

I'm looking for some help with resolving unnecessary contact forces appearing between links in the model.

Below attached is the SDF format code exported directly from Fusion 360 containing all links and joints:

<details>
  <summary>SDF for robot</summary>

```XML
<?xml version="1.0" ?>
<sdf version="1.6">
	<model name="rehab_v2">
		<link name="End_Effector">
			<self_collide>false</self_collide>
			<pose>0.034099980491762015 0.7651576256927808 0.867691695573164 0.0 -0.0 0.0</pose>
			<inertial>
				<pose>0.0 0.0 0.0 0 0 0</pose>
				<mass>1.3416403685863199</mass>
				<inertia>
					<ixx>0.01408292135767879</ixx>
					<ixy>-0.00011048307712142106</ixy>
					<ixz>0.0005551319989724335</ixz>
					<iyy>0.015567706923254992</iyy>
					<iyz>0.002569151816453814</iyz>
					<izz>0.0029787361109184533</izz>
				</inertia>
			</inertial>
			<collision name="End_Effector_collision">
				<geometry>
					<mesh>
						<uri>/home/aamod/Stoodies/wslfiles/DevelopmentEnv/MRProject/Sony/SDF2MJCF/rehab_v2/meshes/CAD/End_Effector.stl</uri>
						<scale>0.001 0.001 0.001</scale>
					</mesh>
				</geometry>
			</collision>
			<visual name="End_Effector_visual">
				<geometry>
					<mesh>
						<uri>/home/aamod/Stoodies/wslfiles/DevelopmentEnv/MRProject/Sony/SDF2MJCF/rehab_v2/meshes/CAD/End_Effector.stl</uri>
						<scale>0.001 0.001 0.001</scale>
					</mesh>
				</geometry>
			</visual>
		</link>
		<link name="Base">
			<self_collide>false</self_collide>
			<pose>0.0010865033095196616 0.19236500102091017 0.31820753236320615 0.0 -0.0 0.0</pose>
			<inertial>
				<pose>0.0 0.0 0.0 0 0 0</pose>
				<mass>90.10842913836647</mass>
				<inertia>
					<ixx>13.440162201021668</ixx>
					<ixy>0.0020396850434465366</ixy>
					<ixz>0.0021381626864994305</ixz>
					<iyy>9.865328750006217</iyy>
					<iyz>-1.0740804259069854</iyz>
					<izz>4.924014808682245</izz>
				</inertia>
			</inertial>
			<collision name="Base_collision">
				<geometry>
					<mesh>
						<uri>/home/aamod/Stoodies/wslfiles/DevelopmentEnv/MRProject/Sony/SDF2MJCF/rehab_v2/meshes/CAD/Base.stl</uri>
						<scale>0.001 0.001 0.001</scale>
					</mesh>
				</geometry>
			</collision>
[... 477 lines cut ...]
			<geom name="Glider_collision" class="unused" type="mesh" group="3" friction="1 0.0050000000000000001 0.0001" pos="0 0 0" euler="0 -0 0" mesh="Glider"/>
			<geom name="Glider_visual" class="unused" type="mesh" contype="0" conaffinity="0" group="0" pos="0 0 0" euler="0 -0 0" mesh="Glider"/>
			<joint name="Slider1" class="unused" type="slide" pos="1.8038723621230379e-05 -2.0931080508823641e-06 -0.12155235075527027" axis="-0.0001485178423013016 -5.6214670093840648e-08 0.99999998897122366" limited="true" range="-0.20000000000000001 0.39000000000000001"/>
			<body name="Ext_Holder" pos="-5.2592823887608921e-07 -0.0041450422041162671 0.02456458809208184" euler="0 -0 0">
				<inertial pos="0 0 0" mass="0.59422050168646334" fullinertia="0.00029118736875434481 0.00044525401105720448 0.00035855008037949577 -1.1925661837612901e-07 -2.3398724991885446e-08 1.7691439615600759e-05"/>
				<geom name="Ext_Holder_collision" class="unused" type="mesh" group="3" friction="1 0.0050000000000000001 0.0001" pos="0 0 0" euler="0 -0 0" mesh="Ext_Holder"/>
				<geom name="Ext_Holder_visual" class="unused" type="mesh" contype="0" conaffinity="0" group="0" pos="0 0 0" euler="0 -0 0" mesh="Ext_Holder"/>
				<joint name="Revolute3" class="unused" type="hinge" pos="-7.8427560669738772e-11 0.0041429420692316588 -0.021116940225950009" axis="-0.9999997024401851 -0.00075700858113645654 -0.00014851784230131182" limited="true" range="0 360"/>
			</body>
		</body>
	</body>
	</worldbody>
	<equality>
		<connect name="equality_constraint" active="true" body1="Ext_Cover" body2="Ext_Holder" anchor="0 0 0" />
	</equality>
</mujoco>
```

</details>

Here is a video, illustrating the model simulation with the equality constraint enabled:
[Screencast from 07-11-23 03:56:29 PM IST.webm](https://github.com/google-deepmind/mujoco/assets/77502673/9cfd4941-cf66-49b7-8c91-922dad91f498)

And here is a video with the equality constraints commented out:
[Screencast from 07-11-23 04:00:29 PM IST.webm](https://github.com/google-deepmind/mujoco/assets/77502673/eaf75515-282b-4a22-820f-f25fe732925c)

nimrod-gileadi, 2023-11-07:
Are those collisions between `Ext_Cover` and `Ext_Holder`? If so, you can specifically exclude those collisions with an `<exclude>` tag: https://mujoco.readthedocs.io/en/latest/XMLreference.html#contact-exclude

```xml
<contact>
  <exclude body1="Ext_Cover" body2="Ext_Holder" />
</contact>
```

aamodbk, 2023-11-08:
> Are those collisions between `Ext_Cover` and `Ext_Holder`? If so, you can specifically exclude those collisions with an `<exclude>` tag: https://mujoco.readthedocs.io/en/latest/XMLreference.html#contact-exclude
> 
> ```
> <contact>
>   <exclude body1="Ext_Cover" body2="Ext_Holder" />
> </contact>
> ```

I tried adding the contact and exclude tag inside it to exclude those collisions but it is still showing those contact points and contact forces. What might be the issue here?

Balint-H, 2023-11-08:
You can enable the labelling of the contact points in the interactive viewer (under the rendering options). That will show you which geoms are colliding, so you can exclude them.

kevinzakka, 2024-01-02:
Marking as resolved, feel free to re-open this if you weren't able to solve your issue.