google-deepmind/mujoco #1175: How can I set the default joint angles?

BolunDai0216, 2023-11-08:
Hi,

I'm wondering how can I set the default joint angles such that when running the passive viewer and pressing the reset button, the robot will go to the default joint angles instead of the all zero joint configuration.

Thanks in advance!

yuvaltassa, 2023-11-08:
The reset button does go to the default configuration, so I'm not sure what you are asking?

Perhaps you want to load a keyframe?

BolunDai0216, 2023-11-08:
Yes, I want to set the robot back to some nominal configuration.

yuvaltassa, 2023-11-09:
The default configuration is not all zeros. It is the initial configuration, which is what you get when you hit Backspace in the viewer window. If you want a different configuration you can just write whatever you want into `data.qpos`.

Does this answer your question?

yuvaltassa, 2023-11-10:
Ah, now that is a a nice ask!

A better way to phrase your request would be "can you please add at least one button (or allow the user to add buttons) whose state can be read from Python and be used to trigger user code". You would use it for resets, somebody else could use it to change the color of a geom, etc. Am I right?

I think someone asked for this before. I guess the problem is that this is a slippery slope. You want a button, somebody else wants a slider, somebody else wants a whole new custom UI section.

One could imagine exposing the [mjuiState](https://mujoco.readthedocs.io/en/latest/APIreference/APItypes.html#mjuistate) and giving people example code for how to add UI elements and read their state.

I also think that for your very specific use case one could enable the key loading behaviour in the passive viewer. But passing some sort of UI state to Python seems like the more generally useful thing.

@saran-t, opinions?

BolunDai0216, 2023-11-10:
@yuvaltassa Yes, that is exactly something that I am looking for!

Balint-H, 2023-11-10:
Is the `springref` parameter not relevant in this case? You could set a stiffness and damping, and have it be passively attracted to that non-zero qpos  state. See:
https://mujoco.readthedocs.io/en/stable/XMLreference.html#body-joint-springref

Balint-H, 2023-11-10:
An alternate temporary option would be to set the springref parameter which will passively attract the model to your non-zero qpos state.  This of course will have a transient period where your model will take some time before it arrives at its desired pose. This will also alter equilibrium pose so it may not necessarily be what you need.

Keyframes could also get you to set the state to a configuration you like.  Until a custom UI button option is implemented, it may be your best bet. See how the humanoid.xml file implements it. You can select the keyframe with "Key" slider under the simulation options, then use the "Load key" button to teleport your model to it. If you are having trouble getting the keyframe, pause the simulation, then use the "Join" sliders on the right hand side to set up your desired pose, then press "Copy pose" to get it to embed in your xml.

yuvaltassa, 2023-11-12:
I'm about to push a change that enables keyframe loading in the passive viewer in the next couple of days which will solve the OP's initial request.

However some sort of user-access to the GUI state, in particular to custom GUI elements still seems like a very valid ask.