google-deepmind/mujoco #1178: How are contact points determined during convex-convex collision?

Chuanfang-Neptune, 2023-11-09:
Hi,

I'm a student working with MuJoCo for neuromechanical simulation. I'm currently building some animal models and trying to understand the collision detection theory, because it would affect how the force applys and how the animal behaves during walking etc.

I'm a bit confused by how the contact points are determined between pairs of meshes. 

I saw in the [MuJoCo paper](https://ieeexplore.ieee.org/stamp/stamp.jsp?tp=&arnumber=6386109) that "_Collision detection uses dedicated pair-wise functions when possible, and otherwise defaults to a general-purpose convex collider (implemented by libccd)_." 

However, [libccd](https://github.com/danfis/libccd) seems to only check if two convex shapes intersect but does not give a contact point or normal where the collision force should apply. I also found the flag `multiccd` could let the convex-convex collider find multiple contact points but it was not explained how the points were located. 

Could you please give some more info on how exactly the contact points are determined between convex objects?

nimrod-gileadi, 2023-11-09:
For Convex-convex collisions, CCD does give a contact point, with the function `ccdMPRPenetration`: https://github.com/danfis/libccd/blob/master/src/ccd/ccd.h#L146

Multiccd runs the same collision code 4 more times, with the objects slightly rotated around the contact point.

Chuanfang-Neptune, 2023-11-09:
Thank you for your pointer. I would take a closer look at how it works

closed 2023-11-09 by Chuanfang-Neptune