google-deepmind/mujoco #1179: In Development Simulator Features

reginald-mclean, 2023-11-09:
Hi,
Thank you for maintaining and updating the Mujoco bindings. 

I am the project manager of [Meta-World](https://github.com/Farama-Foundation/Metaworld/). We are doing an exploration of updating Meta-World to use the MJX features. Is there any timeline for the features that are listed under "In Development" in the docs? 

Thanks in advance for your help.

erikfrey, 2023-11-09:
Hello!  The short answer is no - we have a prioritization for in-development features, but we have not estimated the work involved just yet.

That said, I can tell you the current top of the stack is:

1) Tendons
2) Muscles
3) More sparsification (of Jacobians and other matrices)

But we're open to hearing from folks - what features would help you in your project?

reginald-mclean, 2023-11-10:
Hi @erikfrey! Thanks for your reply. I have had to update my XMLs for a few different things:
- Cylinder Geom
- Elliptic Cone
- Condim=4

Once I (naively) make changes to the XMLs based on what is supported, I have an issue with collisions between plane geoms and cylinder geoms not being implemented.

yuvaltassa, 2024-01-02:
Is this all sorted?

Closing this issue, if there are missing items in the Feature Parity section, please send us a PR to update it.

Thanks

closed 2024-01-02 by yuvaltassa

jamesheald, 2024-03-30:
Hey @erikfrey. Regarding tendons and muscles, do you have a rough idea of when these will be supported?

vikashplus, 2024-03-30:
+1 Any update on 
  1. Tendons
  2. Ellipsoid collisions?