google-deepmind/mujoco #1180: How to make the passive view use another camera?

BolunDai0216, 2023-11-09:
Hi,

I have created a fixed camera using

```xml
<body name="fixed_camera" pos="0 -3.0 0.8">
    <camera name="camera" mode="fixed" euler="1.5708 0 0" fovy="60"/>
</body>
```

which should have camera id 0. I was wondering how can I make the passive viewer render the scene using the view obtained from this camera?

Thanks in advance!

yuvaltassa, 2023-11-10:
The Rendering tab has a "Camera" drop down menu.
You can also use the `[` and `]` keys to cycle the fixed cameras (and `Esc` to go back to the free camera).
If you click on the Help button (`F1` key), you will see this documented in the viewer.

closed 2023-11-10 by yuvaltassa

BolunDai0216, 2023-11-10:
@yuvaltassa Thanks for the response. I was referring to how this can be achieved using the Python bindings.

BolunDai0216, 2023-11-10:
I was not able to use another camera (like switching from the camera with ID -1 to the one with ID 1). But I did figure out how to change the initial viewpoint of the default free camera:

```python
viewer = mujoco.viewer.launch_passive(model, data)
viewer.cam.distance = 3.0
viewer.cam.azimuth = 90
viewer.cam.elevation = -45
viewer.cam.lookat[:] = np.array([0.0, -0.25, 0.824])
```

yuvaltassa, 2023-11-12:
You can also do the above in your XML by changing `<statistic {center, extent}/>` and `<visual> <global {azimuth, elevation}/> </visual>`. See [XML reference](https://mujoco.readthedocs.io/en/latest/XMLreference.html) for details.

I'm not sure switching cameras programmatically makes sense for the passive viewer. This is an interactive tool meant for user interaction. If you want procedural controls, use the offline rendering tools.