google-deepmind/mujoco #1185: GLContext: incorrect free when mediapy imported

tmplt, 2023-11-11:
I've stumbled upon a confusing error when `mediapy` is imported and an renderer instantiated, MWE:
```bash
$ python -c "import mediapy; import mujoco; model = mujoco.MjModel.from_xml_string('<mujoco/>'); renderer = mujoco.Renderer(model)"
Exception ignored in: <function GLContext.__del__ at 0x7f215f6a4b80>
Traceback (most recent call last):
  File "/nix/store/r325k7kv6ra6yh74dvh4x3g1xl692ma3-python3.11-mujoco-3.0.0/lib/python3.11/site-packages/mujoco/glfw/__init__.py", line 41, in __del__
  File "/nix/store/r325k7kv6ra6yh74dvh4x3g1xl692ma3-python3.11-mujoco-3.0.0/lib/python3.11/site-packages/mujoco/glfw/__init__.py", line 35, in free
TypeError: 'NoneType' object is not callable
```

Removing the `mediapy` import, or simply not assigning the renderer to a variable fixes it. 

I experience this with v3.0.0 in an in-progess package for NixOS (https://github.com/NixOS/nixpkgs/pull/266099). Trying to figure out where this error originates. `mediapy` is version 1.1.9. Python is 3.11.

kevinzakka, 2023-11-22:
Can reproduce this on my end as well on Ubuntu and Mac.

kevinzakka, 2023-12-08:
We're looking into this, will have an update soon.

kevinzakka, 2023-12-15:
Hi @alberthli and @tmplt, can you check if the latest release (3.1.0) fixes your issue? We've added a `close` method that you should call to free up resources associated with the renderer object. You can use it as follows:

```python
# Option 1.
renderer = Renderer(...)
# Use renderer.
renderer.close()

# Option 2.
with Renderer(...) as renderer:
# Use renderer.
```

alberthli, 2023-12-15:
Hi @kevinzakka, I can confirm it works (I only tested option 1).

kevinzakka, 2023-12-15:
Awesome, thanks @alberthli. Marking as resolved then.

closed 2023-12-15 by kevinzakka

tmplt, 2023-12-16:
Explicitly calling `renderer.close()` or `del rendered` does resolve it on 3.1.0 for me, but omitting it yields errors. But perhaps `__del__` isn't invoked on the OP example?

hqm666, 2024-06-02:
> Hi @alberthli and @tmplt, can you check if the latest release (3.1.0) fixes your issue? We've added a `close` method that you should call to free up resources associated with the renderer object. [...]

hello @kevinzakka , this method doesn't work. My code is like this:
...
import mediapy as media
...
renderer = mujoco.Renderer(model)
...
renderer.close()

When i run it, the problem reappears agian:
Exception ignored in: <function Renderer.__del__ at 0x000001D063353CE0>
TypeError: 'NoneType' object is not callable

zxhuang97, 2024-07-06:
Hi, I still have the same issue on 3.1.6 when I have a persistent renderer in my env class. Am I supposed to create it on the fly using the with statement? I thought this would slow down the rendering.
Thanks.