google-deepmind/mujoco #1186: Python: redefining renderer corrupts its future renders

tmplt, 2023-11-11:
```python
>>> import mujoco
>>> import mediapy as media
>>> model = mujoco.MjModel.from_xml_string('<mujoco/>'); data = mujoco.MjData(model); renderer = mujoco.Renderer(model);
>>> media.write_image('image.png', renderer.render()) # observe expected black image
>>> renderer = mujoco.Renderer(model);
>>> media.write_image('image.png', renderer.render()) # observe corrupted image
>>> media.write_image('image.png', renderer.render()) # observe differently corrupted image
```
or alternatively
```python
>>> import numpy as np
>>> # before renderer redefine
>>> np.count_nonzero(renderer.render())
0
>>> renderer = mujoco.Renderer(model);
>>> np.count_nonzero(renderer.render())
42109
>>> np.count_nonzero(renderer.render())
42129
```

Python bindings from https://github.com/NixOS/nixpkgs/pull/266099 (v3.0.0).

tmplt, 2023-11-11:
I'm hazarding a guess that something funky happens here on the second call: https://github.com/google-deepmind/mujoco/blob/d0cabcf7dbf004f234bc0d9ba9dfe719a97d4e66/python/mujoco/renderer.py#L79-L89

tmplt, 2023-11-13:
Relevant pytest: https://github.com/tmplt/mujoco/commit/8c067f06b09314ad69e5692e5ea8830c3041f0c1.

saran-t, 2023-11-16:
Can you try this? https://github.com/saran-t/mujoco/commit/25749621593757d1aff9b9732c52c193374b7b40

tmplt, 2023-11-16:
No effect. I can reproduce with the patch applied, but also with `MUJOCO_GL=glfw`.

kevinzakka, 2023-12-15:
Hi @tmplt, can you check if the latest release (3.1.0) fixes your issue? We've added a `close` method that you should call to free up resources associated with the renderer object. You can use it as follows:

```python
# Option 1.
renderer = Renderer(...)
# Use renderer.
renderer.close()

# Option 2.
with Renderer(...) as renderer:
# Use renderer.
```

tmplt, 2023-12-16:
@kevinzakka with 3.1.0:
```python
 python -c "import mujoco; import numpy as np; model = mujoco.MjModel.from_xml_string('<mujoco/>'); data = mujoco.MjData(model); renderer = mujoco.Renderer(model); before = np.count_nonzero(renderer.render()); renderer.close(); renderer = mujoco.Renderer(model); after = np.count_nonzero(renderer.render()); print([before, after])"
[189296, 84945]
```
Both are expected to be zero.

I tried to repro the OP in 3.0.1 and 3.0.0 but was unable to: the first render yields non-zero also. I must have hit some edge-case where it does return zero during experimentation.

kevinzakka, 2024-01-02:
I have a fix incoming that will make the next release, but if you want an immediate fix, add the following 3 lines after the gl context free [here](https://github.com/google-deepmind/mujoco/blob/main/python/mujoco/renderer.py#L317):

```python
if self._mjr_context:
  self._mjr_context.free()
self._mjr_context = None
```

commit 1c7809e5 (2024-01-02): Free `MjrContext` in Renderer destructor. Fixes #1186.

PiperOrigin-RevId: 595125644
Change-Id: I5dadc3508ec296ba8844fed2a8f5d2de7e53221d
closed 2024-01-02 by copybara-service[bot]