google-deepmind/mujoco #1188: How to contribute to mujoco_mjx.

jotix16, 2023-11-12:
**Is there a simple approach for contributing to mujoco_mjc?**

Since mujoco_mjx is a python library, I was wondering whether it could be installed in editor mode `pip install -e ..`?

kevinzakka, 2023-11-13:
> I was wondering whether it could be installed in editor mode pip install -e .

Yes

jotix16, 2023-11-13:
Hi @kevinzakka,

What would be the steps to do that? Clone the mujoco repo, build with cmake in build/, ..

Where/when do I call `pip install -e …`?

jotix16, 2023-11-14:
I was confused by the absence of `setup.py` which is supposedly required for `pip install -e .`, but apperently `pyproject.toml` is enough.

The steps I followed:

1. Install the whole mujoco as you would normally do:
``` bash
pip install mujoco
```
2. Clone the whole mujoco repo:
``` bash
git clone https://github.com/google-deepmind/mujoco.git
```
3. Install mjx in development move
``` bash
cd mujoco/mjx
pip install -e .
```

I am closing the issue. 

PS: I could open a PR for the README if this info would help others who want to install from source.

closed 2023-11-14 by jotix16