google-deepmind/mujoco #1189: Custom XML/URDF pipeline

dibbla, 2023-11-13:
Hi,

I'm a student and I'm trying to use MuJoCo for custom RL environment for custom robot. I was using SolidWorks for modeling and exporting things into Mujujo for RL training.

I'm looking for some advices in working in this pipeline. I have created a simple wheeled leg robot in SolidWorks and export them with sw2urdf extension from ROS [link](https://wiki.ros.org/sw_urdf_exporter). However, although my robot seems good in SolidWorks, there were some joints that slipped out.

My exported assets can be found [here](https://drive.google.com/drive/folders/1xcBCoqocaM4FNglstmKCxgfUw3_X7Px4?usp=sharing). This folder is the direct output of ROS sw2urdf extension.  I just launch the viewer and drag the urdf into the viewer. The result is kind of hilarious.

[image]

I'm new to robotics, any suggestions to this issue/pipeline would be appreciated.

yuvaltassa, 2023-11-13:
Can you please load your URDF in something else to confirm that the issue is with MuJoCo URDF import rather than `sw2urdf` ?

dibbla, 2023-11-13:
> Can you please load your URDF in something else to confirm that the issue is with MuJoCo URDF import rather than `sw2urdf` ?

Hi Yuval! 

Thanks for the fast reply! I found indeed the sw2urdf broken. The key is to, not only create the reference axis in SolidWorks, but also create the corresponding coordinate system. That solves the problem.

closed 2023-11-13 by dibbla