google-deepmind/mujoco #1190: MuJoCo 3.0.0 python bindings: viewer.user_scn.flags are not respected in the actual rendered viewer

lksmac, 2023-11-13:
Hi,

after upgrading to MuJoCo 3.0.0 (python bindings), changes I make to the `viewer.user_scn.flags` are not respected in the actual rendered viewer scene anymore.

Example:
```
viewer = launch_passive(m, d)
viewer.lock()
viewer.user_scn.flags[mujoco.mjtRndFlag.mjRND_SHADOW] = 0
viewer.user_scn.flags[mujoco.mjtRndFlag.mjRND_REFLECTION] = 0
viewer.sync()
```

The [documentation](https://mujoco.readthedocs.io/en/3.0.0/python.html#pyviewerpassive) states that:
"user_scn: an mjvScene object that allows users to add custom visualization geoms to the rendered scene. This is separate from the mjvScene that the viewer uses internally to render the final scene [...]"

My question: **How is it possible to change the flags of the internal mjvScene that the viewer uses for rendering?**

Thanks for the help.

vikashplus, 2024-01-11:
I am programmatically trying to enable/disable Fog by changing the `viewer.user_scn.flags` which is also not working. I believe this is aligned with this thread's request.

closed 2024-02-02 by copybara-service[bot]

vikashplus, 2024-02-02:
Thanks @saran-t for this feature :)

commit 7bb0ce42 (2024-02-02): [Python Viewer] Allow users to override rendering flags via `user_scn`.

Fixes #1190

PiperOrigin-RevId: 603672602
Change-Id: I3ff1bb0dca7e913b36f1b241aaea8b8bb95cadff