google-deepmind/mujoco #1191: [Unity Plugin] MissingReferenceException: The object of type 'Mesh' has been destroyed but you are still trying to access it.

ganyuanzhen, 2023-11-13:
Hi,

I'm a student and I'm trying to use MuJoCo 2.3.1 to implement a simple parallel robot using the Unity plugin 2.3.1 and Unity version 2020.3.48f1c1 on Windows 11.

I'm looking for some help with the problem that the following errors will occur after hitting the play button.

I imported the plugin via git using the same URL in the https://github.com/Balint-H/mj-unity-tutorial repo.

The model, which is provided below, works well in the `mj-unity-tutorial`. But when I tried to import it to my project, some errors will occor.

```
MissingReferenceException: The object of type 'Mesh' has been destroyed but you are still trying to access it.
Your script should either check if it is null or you should not destroy the object.
UnityEngine.Mesh.GetAllocArrayFromChannel[T] (UnityEngine.Rendering.VertexAttribute channel, UnityEngine.Rendering.VertexAttributeFormat format, System.Int32 dim) (at <90585cf0c5a24f279207485a8de593e7>:0)
UnityEngine.Mesh.GetAllocArrayFromChannel[T] (UnityEngine.Rendering.VertexAttribute channel) (at <90585cf0c5a24f279207485a8de593e7>:0)
UnityEngine.Mesh.get_vertices () (at <90585cf0c5a24f279207485a8de593e7>:0)
Mujoco.MjcfGenerationContext.GenerateMeshMjcf (UnityEngine.Mesh mesh, System.Xml.XmlElement mjcf) (at Library/PackageCache/org.mujoco@d566e5ebe4/Runtime/Tools/MjcfGenerationContext.cs:93)
Mujoco.MjcfGenerationContext.GenerateAssetsMjcf (System.Xml.XmlElement mjcf) (at Library/PackageCache/org.mujoco@d566e5ebe4/Runtime/Tools/MjcfGenerationContext.cs:87)
Mujoco.MjcfGenerationContext.GenerateMjcf (System.Xml.XmlElement mjcf) (at Library/PackageCache/org.mujoco@d566e5ebe4/Runtime/Tools/MjcfGenerationContext.cs:40)
Mujoco.MjScene.GenerateSceneMjcf (System.Collections.Generic.IEnumerable`1[T] components) (at Library/PackageCache/org.mujoco@d566e5ebe4/Runtime/Components/MjScene.cs:416)
Mujoco.MjScene.CreateScene (System.Boolean skipCompile) (at Library/PackageCache/org.mujoco@d566e5ebe4/Runtime/Components/MjScene.cs:147)
UnityEngine.Debug:LogException(Exception)
Mujoco.MjScene:CreateScene(Boolean) (at Library/PackageCache/org.mujoco@d566e5ebe4/Runtime/Components/MjScene.cs:150)
Mujoco.MjScene:Start() (at Library/PackageCache/org.mujoco@d566e5ebe4/Runtime/Components/MjScene.cs:86)

MissingReferenceException: The object of type 'Mesh' has been destroyed but you are still trying to access it.
Your script should either check if it is null or you should not destroy the object.
UnityEngine.Mesh.GetAllocArrayFromChannel[T] (UnityEngine.Rendering.VertexAttribute channel, UnityEngine.Rendering.VertexAttributeFormat format, System.Int32 dim) (at <90585cf0c5a24f279207485a8de593e7>:0)
UnityEngine.Mesh.GetAllocArrayFromChannel[T] (UnityEngine.Rendering.VertexAttribute channel) (at <90585cf0c5a24f279207485a8de593e7>:0)
UnityEngine.Mesh.get_vertices () (at <90585cf0c5a24f279207485a8de593e7>:0)
Mujoco.MjcfGenerationContext.GenerateMeshMjcf (UnityEngine.Mesh mesh, System.Xml.XmlElement mjcf) (at Library/PackageCache/org.mujoco@d566e5ebe4/Runtime/Tools/MjcfGenerationContext.cs:93)
Mujoco.MjcfGenerationContext.GenerateAssetsMjcf (System.Xml.XmlElement mjcf) (at Library/PackageCache/org.mujoco@d566e5ebe4/Runtime/Tools/MjcfGenerationContext.cs:87)
Mujoco.MjcfGenerationContext.GenerateMjcf (System.Xml.XmlElement mjcf) (at Library/PackageCache/org.mujoco@d566e5ebe4/Runtime/Tools/MjcfGenerationContext.cs:40)
Mujoco.MjScene.GenerateSceneMjcf (System.Collections.Generic.IEnumerable`1[T] components) (at Library/PackageCache/org.mujoco@d566e5ebe4/Runtime/Components/MjScene.cs:416)
Mujoco.MjScene.CreateScene (System.Boolean skipCompile) (at Library/PackageCache/org.mujoco@d566e5ebe4/Runtime/Components/MjScene.cs:156)
Mujoco.MjScene.Start () (at Library/PackageCache/org.mujoco@d566e5ebe4/Runtime/Components/MjScene.cs:86)

NullReferenceException: Failed to create Mujoco runtime.
Mujoco.MjScene.StepScene () (at Library/PackageCache/org.mujoco@d566e5ebe4/Runtime/Components/MjScene.cs:321)
Mujoco.MjScene.FixedUpdate () (at Library/PackageCache/org.mujoco@d566e5ebe4/Runtime/Components/MjScene.cs:95)

```

Here is a model which I used,
```XML
<mujoco>
	<default>
		<joint type="hinge" frictionloss="0.1" damping="2" />
		<position ctrllimited="true" ctrlrange="-3.14 3.14" forcelimited="true" forcerange="-50 50" kp="10" />
	</default>
	<asset>
		<texture type="skybox" builtin="gradient" rgb1="0.3 0.5 0.7" rgb2="0 0 0" width="512" height="512" />
		<texture name="texplane" type="2d" builtin="checker" rgb1=".2 .3 .4" rgb2=".1 0.15 0.2" width="512" height="512" mark="cross" markrgb=".8 .8 .8" />
		<material name="matplane" reflectance="0.3" texture="texplane" texrepeat="1 1" texuniform="true" />
	</asset>
	<worldbody>
		<light directional="true" diffuse=".4 .4 .4" specular="0.1 0.1 0.1" pos="0 0 5.0" dir="0 0 -1" castshadow="false" />
		<light directional="true" diffuse=".6 .6 .6" specular="0.2 0.2 0.2" pos="0 0 4" dir="0 0 -1" />
		<geom name="ground" type="plane" size="0 0 1" pos="0 0 0" quat="1 0 0 0" material="matplane" condim="1" />

		<body name="b0" pos="0 0 1">
			<geom type="box" size="0.2 0.025 0.025" rgba=".8 .2 .1 1" />

			<body name="b1" pos="-0.2 0 -0.2" euler="0 90 0">
				<geom type="box" size="0.2 0.025 0.025" rgba=".8 .2 .1 1" />
				<joint name="j0" type="hinge" pos="-0.2 0 0" axis="0 1 0" />
			</body>

			<body name="b2" pos="0.2 0 -0.2" euler="0 90 0">
				<geom type="box" size="0.2 0.025 0.025" rgba=".8 .2 .1 1" />
				<joint name="j2" type="hinge" pos="-0.2 0 0" axis="0 1 0" />
			</body>
		</body>

		<body name="b3" pos="0 0 0.6" euler="0 0 0">
			<freejoint />
			<geom type="box" size="0.2 0.025 0.025" rgba=".8 .2 .1 1" />
		</body>
	</worldbody>
	<contact>
		<exclude name="e0" body1="b0" body2="b1" />
		<exclude name="e1" body1="b0" body2="b2" />
		<exclude name="e2" body1="b3" body2="b2" />
		<exclude name="e3" body1="b3" body2="b1" />
	</contact>
	<equality>
		<connect name="c1" body1="b2" body2="b3" anchor="0.2 0 0" />
		<connect name="c2" body1="b1" body2="b3" anchor="0.2 0 0" />
	</equality>
	<actuator>
		<position joint="j0" ctrllimited="true" ctrlrange="-0.875 0.875" />
		<position joint="j2" ctrllimited="true" ctrlrange="-0.875 0.875" />
	</actuator>
</mujoco>
```

As expected, the model above works perfectly in the simulate.exe ,

[image]

There are some debug info, which may be helpful,

[image]

[image]

[image]

P.S. I have some other Unity packages installed. But I'm not sure whether they will affect the MuJoCo plugin.

[image]

I'd appreciate it if anyone has time to help me.

Balint-H, 2023-11-13:
Hello! Could you clarify a few things for me?

Did you move your scene to a new project, or did you reimport it at the new project? Did you get the error message when trying to import the model?

ganyuanzhen, 2023-11-14:
> Hello! Could you clarify a few things for me?
> 
> Did you move your scene to a new project, or did you reimport it at the new project? Did you get the error message when trying to import the model?
> 
> I tested it on the MuJoCo 2.3.7 version of the Unity plugin, it imports and runs well. Hopefully its just an issue of the mesh files getting lost when moving a scene from one project or another, in which case a reimport could solve it.
> 
> You could also try upgrading to 2.3.7 (e.g., by using `"org.mujoco": "https://github.com/deepmind/mujoco.git?path=unity#2.3.7"` in your manifest, and not forgetting to upgrade the DLL).

Thank you for the reply! As you mentioned above, I created a new Project and scene, importing all the packaged and scripts I used and I works! 

This is probably an issue when I move my project :)

closed 2023-11-14 by ganyuanzhen