google-deepmind/mujoco #1193: Passive viewer segmentation fault flexcomp element

giorgionicola, 2023-11-14:
I am starting to simulate deformable objects with Mujoco 3.0. Anytime I use model that includes flexcomp the viewer immediately crashes after loading the model, returning a segmentation fault. Models that do not use the flexcomp element works.

The python script used to simulate is the following, directly taken from the python bindings simulation
```
import time`
import mujoco
import mujoco.viewer

m = mujoco.MjModel.from_xml_path('model.xml')
d = mujoco.MjData(m)

with mujoco.viewer.launch_passive(m, d) as viewer:
    start = time.time()
    while viewer.is_running():
        step_start = time.time()
        mujoco.mj_step(m, d)
        with viewer.lock():
            viewer.opt.flags[mujoco.mjtVisFlag.mjVIS_CONTACTPOINT] = int(d.time % 2)
        viewer.sync()
        time_until_next_step = m.opt.timestep - (time.time() - step_start)
        if time_until_next_step > 0:
            time.sleep(time_until_next_step)
```
as models I have tested the following https://github.com/google-deepmind/mujoco/tree/main/model/plugin/elasticity 

From my understanding the segmantation faults happen at:
`with mujoco.viewer.launch_passive(m, d) as viewer:`

since I was able to run multiple simulation steps when the viewer was not launched.

System info
OS: Ubuntu 22.04
Mujoco version 3.0
Official pythjon byndings
Python 3.10

giorgionicola, 2023-11-23:
I am reopening the issue since with models that uses the plugin "mujoco.elasticity.solid" the viewer crashes

quagla, 2023-11-24:
Are you compiling mujoco from source or using the release?

giorgionicola, 2023-11-24:
I am using the latest release 3.0.1

commit e3b3f963 (2023-11-15): Add `flex_texcoordadr` to `mjvSceneState`. Fixes #1193.

PiperOrigin-RevId: 582630943
Change-Id: I113b135752268773e0afafe004b8f27c1e96ffc7
commit 8010aad7 (2023-11-27): Add `flex_elemlayer` to `mjvSceneState`. Fixes #1193 and #1219.

PiperOrigin-RevId: 585581657
Change-Id: Iec4da74437544184c19765a2feee1cca4c368d34
closed 2023-11-27 by copybara-service[bot]