google-deepmind/mujoco #1196: Building MuJoCo from source with debug flags

VasilyRakche, 2023-11-14:
Dear MuJoCo team,

How can I compile from source with debug information (so that I can debug the code during execution)

Best

saran-t, 2023-11-21:
Have you tried running CMake in `RelWithDebInfo ` mode?

VasilyRakche, 2023-11-23:
For future reader, a bit more info from [Stackoverflow](https://stackoverflow.com/a/59314670)
> ```
> 1. Release: high optimization level, no debug info, code or asserts.
> 2. Debug: No optimization, asserts enabled, [custom debug (output) code enabled],
>    debug info included in executable (so you can step through the code with a
>    debugger and have address to source-file:line-number translation).
> 3. RelWithDebInfo: optimized, *with* debug info, but no debug (output) code or asserts.
> 4. MinSizeRel: same as Release but optimizing for size rather than speed.
> ```
> In terms of compiler flags that usually means (since these are supported in most cases on all platforms anyway):
> ```
> 1. Release: `-O3 -DNDEBUG`
> 2. Debug: `-O0 -g`
> 3. RelWithDebInfo: `-O2 -g -DNDEBUG`
> 4. MinSizeRel: `-Os -DNDEBUG`
> ```

Example on usage. Modify [step 3.](https://mujoco.readthedocs.io/en/latest/programming/index.html#building-from-source) as follows:  
```
cmake $PATH_TO_CLONED_REPO -DCMAKE_BUILD_TYPE=Debug -DCMAKE_INSTALL_PREFIX=<install_path> -DMUJOCO_BUILD_EXAMPLES=OFF
```

closed 2023-11-23 by VasilyRakche