google-deepmind/mujoco #1197: "WARNING: Nan, Inf or huge value in QPOS/QACC at DOF 0."

DFREANDY, 2023-11-15:
Thanks for your advice! 
I've tried to set the young modulus from "0.01" to "1000". However, the same problem still exists. 
I also decrease the timestep to "0.0001", and set  `equality="true"`. It doesn't help.
As long as I try to use the `mujoco.elasticity.solid` plugin, the warning will appear.
 "WARNING: Nan, Inf or huge value in QPOS/QACC at DOF 0."

_Originally posted by @DFREANDY in https://github.com/google-deepmind/mujoco/issues/1173#issuecomment-1812118111_

quagla, 2023-11-15:
Maybe your cube contains faces that have an inverted orientation?

quagla, 2023-11-15:
Also, if it's a triangle mesh of the surface only and not tetrahedral of the volume, then you should use the membrane plugin instead. See https://github.com/google-deepmind/mujoco/blob/main/plugin/elasticity/README.md

DFREANDY, 2023-11-15:
I‘ve tried different models from STL files, including cube, cylinder, etc. However, all models will cause the same warning. 
I also tried to use the membrane plugin just now. It leads to the same warning.
I want to load the models from files so that it will be easy for me to simulate the deformable objects in MuJoCo. Dose the plugin support models imported from STL files? (I've noticed that most of the flexcomp models in model/plugin/elasticity are created with `type="grid"`.)

quagla, 2023-11-16:
The issue is that the Poisson ratio should be < 0.5. A value exactly equal to 0.5 means an infinite Lamé volumetric parameter https://en.wikipedia.org/wiki/Lam%C3%A9_parameters which is giving you the issue. 

Remember also that `equality="true"` does not turn off automatically the plugin, so if you only want constraints you have to delete the plugin. Also that when you use the `membrane` plugin, the `thickness` attribute is required (otherwise it defaults to zero and you will get zero stiffness).

Loading STF files is possible :)

DFREANDY, 2023-11-16:
Thanks a lot! I've changed the Poisson ratio, add the `thickness` attribute, and the problem is solved!