google-deepmind/mujoco #1198: Error loading the model: XML Error: Schema violation: unrecognized attribute: 'collision'

PluteW, 2023-11-15:
**How to ask for help**

First, read our quick guide to
[asking good questions](https://github.com/google-deepmind/mujoco#asking-questions).
Below is a template for you to use:

Hi,

I'm a student and I'm trying to use MuJoCo for robot simulation.

I'm looking for some help with XML loading.

I have installed the plug-in follow the instruction. But when i test with hammock.xml in release 3.0.0. I got this in Unity: NullReferenceException: Failed to create Mujoco runtime.

I tried this in 2 platform and got the same result:

1. windows 10 , Unity 2021.1, mujuco 3.0.0
2.  ubuntu 18.04, unity 2020.4, mujoco 3.0.0

Here is a screenshot / video, illustrating my question:

[image]

erez-tom, 2023-11-15:
Thanks for flagging!  This is a bug on our side: this feature was removed [here](https://github.com/google-deepmind/mujoco/commit/9cf1f6eba42516bee1a051d193d7f511e3bce932) and should have been also removed from the Unity codebase.  Feel free to send a PR removing this attribute, or I'll push a fix soon.
Attn: @Balint-H , @yuvaltassa

PluteW, 2023-11-16:
Thanks! I will try it right now!

PluteW, 2023-11-16:
I checked the code with the update you mentioned [here](https://github.com/google-deepmind/mujoco/commit/9cf1f6eba42516bee1a051d193d7f511e3bce932), I didn't find `mjtCollision` in `MjBindings.cs`, I think the releated code has been updated.

I also tried with version 3.0.1, updated serveral hours ago, the problem seems shows again:

[image]

When I tried to rebuild with cmake, for some region problem, I got some problem with download the dependencies.

Could you please give more suggestions to fixed this?

erez-tom, 2023-11-16:
I believe [this](https://github.com/google-deepmind/mujoco/commit/9952f1708cacd0a1bdc3bd171e746962b7e5e4db) should fix it.  PTaL and let us know if it's all good.

closed 2023-11-16 by erez-tom

PluteW, 2023-11-18:
Thanks a lot. It works well with some XML like `balloons.xml`, but with `hammock.xml`, there still some problem like this:
[image]