google-deepmind/mujoco #1199: Physics simulation doesn't take effect after playing in Unity

madejiaaha, 2023-11-15:
Hi,

I'm a student and I'm trying to use MuJoCo for cable-simulation.
I successfully imported the MuJoCo plugin into the Unity project and copied the mujoco.dll to the project file at the following address "Assets\MuJoCo\mujoco.dll". I imported the example into the scene and clicked play and it didn't report any errors, but the model in the scene remained stationary.
[image]

I also tried creating a new scene, as shown here, where I set all the values related to Ball's friction with respect to the platform to 1. However, after playing the animation the ball still rolls downwards and does not show any deceleration.
[image]
Import MuJoCo Scene works fine and can add MuJoCo components normally. But the solver doesn't seem to be working. Does anyone know what causes this problem?

i really appreciated
Rongtao

erez-tom, 2023-11-15:
If you grab the MJCF XML and load it in the native application ("simulate"), does the physical behavior look the same?  Can you edit the MJCF to elicit the desired behavior?  I'm basically proposing to decouple Unity from the question of how to manipulate friction in MuJoCo...  I'm closing the issue for now, but please update us on your learning and re-open if you think there's a real bug at play here.

closed 2023-11-15 by erez-tom

Balint-H, 2023-11-15:
Hello! 

For the first scene, do you have any error messages in your console?

For the second, what's the `condim` of your geom? If its less than 6, the friction will not prevent rolling. See:
https://mujoco.readthedocs.io/en/stable/XMLreference.html#body-geom-condim.
However, having all frictional constants be 1 (and condim be 6) is a bit extreme, what kind of situation are you trying to model? Or are you just exploring the constants?

Lastly, ATM there are a couple issues with the 3.0.0 version of the plugin that are being fixed. If your scenes don't run at all, could you try with version 2.3.7 (making sure to downgrade the DLL too)?

madejiaaha, 2023-11-17:
> Hello!
> 
> For the first scene, do you have any error messages in your console?
> 
> For the second, what's the `condim` of your geom? If its less than 6 (3 is the default), the friction will not affect rolling. See: https://mujoco.readthedocs.io/en/stable/XMLreference.html#body-geom-condim. However, having all frictional constants be 1 (and condim be 6) is a bit extreme, what kind of situation are you trying to model? Or are you just exploring the effects of these parameters?
> 
> Lastly, ATM there are a couple issues with the 3.0.0 version of the plugin that are being fixed. If your scenes don't run at all, could you try with version 2.3.7 (making sure to downgrade the DLL too)?

Thanks for your reply!! I tried downgrading to version 2.3.7 (DLL too), still the same problem, it's possible that it's a problem with my Unity that's causing him to not report errors. I tried it on a different device and the error was reported:

IOException: Error loading the model: XML Error: unsigned integer with an optional suffix {K,M,G,T,P,E} is expected in attribute 'memory' (or the size specified is too big)
Element 'size', line 1

Mujoco.MjVfs.LoadXML (System.String filename) (at D:/Mujoco/mujoco-main/unity/Runtime/Tools/MjVfs.cs:83)
Mujoco.MjEngineTool.LoadModelFromString (System.String contents) (at D:/Mujoco/mujoco-main/unity/Runtime/Tools/MjEngineTool.cs:69)
Mujoco.MjScene.CompileScene (System.Xml.XmlDocument mjcf, System.Collections.Generic.IEnumerable`1[T] components) (at D:/Mujoco/mujoco-main/unity/Runtime/Components/MjScene.cs:179)
Mujoco.MjScene.CreateScene (System.Boolean skipCompile) (at D:/Mujoco/mujoco-main/unity/Runtime/Components/MjScene.cs:171)
Mujoco.MjScene.Start () (at D:/Mujoco/mujoco-main/unity/Runtime/Components/MjScene.cs:89)

NullReferenceException: Failed to create Mujoco runtime.
Mujoco.MjScene.StepScene () (at D:/Mujoco/mujoco-main/unity/Runtime/Components/MjScene.cs:345)
Mujoco.MjScene.FixedUpdate () (at D:/Mujoco/mujoco-main/unity/Runtime/Components/MjScene.cs:98)

Followed the prompts in #807 and now the problem is solved！！！