google-deepmind/mujoco #1202: mj_multiRay() using MJX

stm32f303ret6, 2023-11-16:
Hi, i need to cast multiple rays to simulate lidars, typically thousands of rays are casted per seconds so the algorithm must be fast, mj_multiRay() works with MJX? 

mj_multiRay() using CPU is not enough fast, GPU could speed up the simulation a lot

yuvaltassa, 2023-11-22:
Note that while ray casting is not yet supported in MJX (on our to-do list), the ray casting functions are completely thread safe, which can give you a significant speedup on CPU.

Andrew-Luo1, 2024-03-10:
In case this is still helpful, I think mj_multiRay() can now be implemented by using jax.vmap on mjx.ray(). For example, please see this simple [depth camera implementation](https://github.com/Andrew-Luo1/jax_shac/blob/main/vision/2dof_ball.ipynb).

Batching across environments, I'm getting >3 million FPS on 100x100 depth images; or 3.3e10 ray casts per second; on an NVIDIA 3060 Ti GPU. This is in a simple setting; see the depth renderings and ground truth renderings below.

[image]
[image]

kevinzakka, 2024-08-30:
MJX rangefinder was implemented in https://github.com/google-deepmind/mujoco/commit/51733c2a8ac081ed7c44d8e96bb0f8d6d30059f0 and will be part of the next release.

closed 2024-08-30 by kevinzakka