google-deepmind/mujoco #1203: Do`MuJoCo` and `MuJoCo-MJX` have differences because of FP arithmetic

Kallinteris-Andreas, 2023-11-16:
Hi,

I'm a Maintainer of `Farama/Gymnasium/MuJoCo` environments & `Farama/Gymansium-Robotics` and I'm trying to use `mujoco-mjx` in those repos (we are in the prototyping stage).

Should we expect the same behavior across `MuJoCo` and `MuJoCo-MJX`, or should we expect different behavior due to floating point arithmetic ordering (similarly to different MuJoCo versions?).
I am assuming that would be the case, but want to make sure.
Thanks!

yuvaltassa, 2023-11-16:
Yes, but even more so 🙂 

The useful platforms supported by XLA (GPUs, TPUs) generally use single precision (32 bits) floats or lower.

closed 2023-11-16 by yuvaltassa

erikfrey, 2023-11-16:
Calculations will be different, but you can generally expect that policies trained in MJX will transfer to MuJoCo.  We demonstrate this in our [MJX tutorial](https://colab.sandbox.google.com/github/google-deepmind/mujoco/blob/main/mjx/tutorial.ipynb).

If you really want to dive into this, feel free to review our unit tests where we measure the float tolerances between the various sub-calculations of MJX and MuJoCo, e.g. [smooth_test.py](https://github.com/google-deepmind/mujoco/blob/main/mjx/mujoco/mjx/_src/smooth_test.py#L30)