google-deepmind/mujoco #1204: Compile failed with gcc 12.3.1 at release mode

limymy, 2023-11-17:
Mujoco source code compiling will pass at `Debug` mode but fail at `Release` mode. Here is the output:
```
[build] 在函数‘makeLabel’中,
[build]     内联自‘mjv_addGeoms’于 /home/li/bitbot-mujoco-demo/build/_deps/mujoco-src/src/engine/engine_vis_visualize.c:636:11:
[build] /home/li/bitbot-mujoco-demo/build/_deps/mujoco-src/src/engine/engine_vis_visualize.c:64:3: 错误：‘strncpy’ output may be truncated copying 99 bytes from a string of length 99 [-Werror=stringop-truncation]
[build]    64 |   strncpy(label, txt, 99);
[build]       |   ^~~~~~~~~~~~~~~~~~~~~~~
[build] 在函数‘makeLabel’中,
[build]     内联自‘mjv_addGeoms’于 /home/li/bitbot-mujoco-demo/build/_deps/mujoco-src/src/engine/engine_vis_visualize.c:676:11:
[build] /home/li/bitbot-mujoco-demo/build/_deps/mujoco-src/src/engine/engine_vis_visualize.c:64:3: 错误：‘strncpy’ output may be truncated copying 99 bytes from a string of length 99 [-Werror=stringop-truncation]
[build]    64 |   strncpy(label, txt, 99);
[build]       |   ^~~~~~~~~~~~~~~~~~~~~~~
......
[build] 在函数‘bufread’中,
[build]     内联自‘mj_loadModel’于 /home/li/bitbot-mujoco-demo/build/_deps/mujoco-src/src/engine/engine_io.c:778:3:
[build] /home/li/bitbot-mujoco-demo/build/_deps/mujoco-src/src/engine/engine_io.c:377:3: 错误：‘memcpy’ offset [0, 299] is out of the bounds [0, 0] [-Werror=array-bounds]
[build]   377 |   memcpy(dest, (char*)buf + *ptrbuf, num);
[build]       |   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
```
There are two errors. 
- `strncpy` function at
https://github.com/google-deepmind/mujoco/blob/4d2a176a8afecfc1b31b87aeb015602cb2772769/src/engine/engine_vis_visualize.c#L64
It can be fixed in two method. 
  1. Change it to `strncpy(label, txt, 100) ;` to fit
https://github.com/google-deepmind/mujoco/blob/4d2a176a8afecfc1b31b87aeb015602cb2772769/src/engine/engine_vis_visualize.c#L52
  2. Add `-Wno-stringop-truncation` in the compile options.
- 'memcpy' function at
https://github.com/google-deepmind/mujoco/blob/4d2a176a8afecfc1b31b87aeb015602cb2772769/src/engine/engine_io.c#L377
It need to add compile option `-Wno-array-bounds` and `-Wno-stringop-overread`.

saran-t, 2023-11-21:
I'd prefer not to have to turn errors off if possible. Could you see if you can also find a way to fix the `memcpy` error and send us a PR? (In the meantime feel free to also send in the `engine_vis_visualize` fix as a PR too)

Thanks!

limymy, 2023-11-22:
The `memcpy` error is caused by https://github.com/google-deepmind/mujoco/blob/20eeb7508690f7accf6d5d854d85ee4835edcace/src/engine/engine_io.c#L778
And I found a strange thing. It would pass the compiling when I set `getnint()` to a number larger than `80`.
After adding these code,
```
  int ibuf[74];
  bufread(ibuf, sizeof(int)*getnint(), buffer_sz, buffer, &ptrbuf);
```
the compile error became
```
[build] 在函数‘bufread’中,
[build]     内联自‘mj_loadModel’于 /home/li/bitbot-mujoco-demo/build/_deps/mujoco-src/src/engine/engine_io.c:779:3:
[build] /home/li/bitbot-mujoco-demo/build/_deps/mujoco-src/src/engine/engine_io.c:377:3: 错误：‘memcpy’ forming offset [296, 299] is out of the bounds [0, 296] of object ‘ibuf’ with type ‘int[74]’ [-Werror=array-bounds=]
[build]   377 |   memcpy(dest, (char*)buf + *ptrbuf, num);
[build]       |   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
```
Then changed `ibuf[74]` to `ibuf[75]`, the error became
```
[build] 在函数‘bufread’中,
[build]     内联自‘mj_loadModel’于 /home/li/bitbot-mujoco-demo/build/_deps/mujoco-src/src/engine/engine_io.c:779:3:
[build] /home/li/bitbot-mujoco-demo/build/_deps/mujoco-src/src/engine/engine_io.c:377:3: 错误：‘memcpy’ offset [0, 299] is out of the bounds [0, 0] [-Werror=array-bounds=]
[build]   377 |   memcpy(dest, (char*)buf + *ptrbuf, num);
[build]       |   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
```
So I guess the `getnint()` is actual 75.

Next I tried changing https://github.com/google-deepmind/mujoco/blob/20eeb7508690f7accf6d5d854d85ee4835edcace/src/engine/engine_io.c#L712
to `static int ints[256];`, and it compiled successfully.

I didn't find the root cause of this problem. Can you help me?

limymy, 2023-12-23:
> Is this resolved, @limymy?

Sorry, I've now switched to the precompiled gcc 12.3.0, and it's working fine. The issue occurred when I was using the gcc latest (12.3.1) compiled from source. I have uninstalled it now. I'd like to temporarily close this issue.

closed 2023-12-23 by limymy