google-deepmind/mujoco #1207: [MJX] Choice of implementation in solver.py

WangHanSolo, 2023-11-17:
Hi,

I'm a student and I'm trying to use MJX for reinforcement learning.

I'm looking for some help with reverse mode differentiation with MJX. 
<details>
  <summary>python code</summary>

```python
import jax
import mujoco
from mujoco import mjx

model = mujoco.MjModel.from_xml_path("models/humanoid.xml")
mjx_model = mjx.device_put(model)

@jax.vmap
def batched_step(vel):
  mjx_data = mjx.make_data(mjx_model)
  qvel = mjx_data.qvel.at[0].set(vel)
  mjx_data = mjx_data.replace(qvel=qvel)
  pos = mjx.step(mjx_model, mjx_data).qpos[0]
  return pos

vel = jax.numpy.arange(0.0, 1.0, 0.01)
pos = jax.jit(batched_step)(vel)

grad_fun = jax.grad(lambda v : batched_step(v)[0])
g = grad_fun(vel)
print(g)
```

</details>

The above code throws the following error: 
<details>
  <summary>error message</summary>

```
Traceback (most recent call last):
File "blahblah.py", line 21, in <module>
    g = grad_fun(vel)
        ^^^^^^^^^^^^^
ValueError: Reverse-mode differentiation does not work for lax.while_loop or lax.fori_loop with dynamic start/stop values. Try using lax.scan, or using fori_loop with static start/stop.
```

</details>

Looking into solver.py, I noticed that there is a scan implementation [here](https://github.com/google-deepmind/mujoco/blob/4d2a176a8afecfc1b31b87aeb015602cb2772769/mjx/mujoco/mjx/_src/solver.py#L151). I am curious why the solver was implemented with `jax.lax.while_loop` [here](https://github.com/google-deepmind/mujoco/blob/4d2a176a8afecfc1b31b87aeb015602cb2772769/mjx/mujoco/mjx/_src/solver.py#L366) instead of the scan implementation.

I did some comparison between the two and the difference in positions and gradients were trivial.

Thanks for the awesome work you folks are doing!

erikfrey, 2023-11-22:
Duplicate of #1182

erikfrey, 2023-11-22:
Hi @WangHanSolo please check out #1182 - I just provided some details in that issue, let's continue further discussion there.

closed 2023-11-22 by erikfrey