google-deepmind/mujoco #1209: Implementing Contact Force Penalization in MJX: Transitioning from Brax's `pipeline_state.qf_constraint`

bheijden, 2023-11-19:
Hi MuJoCo Team,

I am a PhD student and working on a project involving reinforcement learning. I'm in the process of transitioning a sim2real pusher task from Brax to MJX.

In Brax, I leveraged the `pipeline_state.qf_constraint` feature to integrate penalties for substantial contact forces into my reward function. I am seeking a parallel approach in MJX for achieving the same functionality. Specifically, my goal is to access data on contact forces, including their magnitude and direction in Cartesian coordinates.

Any guidance or suggestions on how to replicate or approximate this feature in MJX would be helpful!

Thank you for your assistance!

btaba, 2023-12-08:
Hi @bheijden 

Indeed `qfrc_constraint` is similar to the one in `brax`. For cartesian contact forces, it looks like you'll want something like `mj_contactForce`, see https://github.com/google-deepmind/mujoco/issues/997#issuecomment-1647992310 which isn't currently implemented in MJX but is available in MuJoCo after a `mjx.get_data`

bheijden, 2023-12-09:
Oke, thanks :)!

closed 2023-12-09 by bheijden