google-deepmind/mujoco #1210: Inconsistency of appearance and physical behavior of a body

DimaRubyAI, 2023-11-20:
When mesh using scale (maybe in some additional cases?) the physical behavior doesn't fit the interference between the bodies.
It's look like some physical attributes are changing by scale (mass distribution) when other attributes stay of unscaled body.

the example is based on official torus.xml example
```
<mujoco>
  <extension>
    <plugin plugin="mujoco.sdf.torus">
      <instance name="torus">
        <config key="radius1" value="0.35"/>
        <config key="radius2" value="0.15"/>
      </instance>
    </plugin>
  </extension>

  <asset>
    <mesh name="torus">
      <plugin instance="torus"/>
    </mesh>
    <mesh name="torus_b" scale="10 10 10">
      <plugin instance="torus"/>
    </mesh>    
  </asset>

  <option sdf_iterations="10" sdf_initpoints="40"/>

  <include file="scene.xml"/>

  <default>
    <geom solref="0.01 1" solimp=".95 .99 .0001" friction="0.1"/>
  </default>

  <worldbody>
    <body pos="-1 0.05 6">
      <freejoint/>
      <geom type="sdf" mesh="torus_b" rgba=".2 .2 .8 1">
        <plugin instance="torus"/>
      </geom>
    </body>
    <body pos="-1 0 3">
      <geom type="sdf" mesh="torus" rgba=".8 .2 .2 1">
        <plugin instance="torus"/>
      </geom>
    </body>
    <light name="left" pos="-2 0 7" cutoff="80"/>
    <light name="right" pos="2 0 7" cutoff="80"/>
  </worldbody>
</mujoco>
```

https://github.com/google-deepmind/mujoco/assets/140307654/ccb2fca9-3fbe-4bc7-a212-ae194d336e7a

for unscaled case the behavior seems reasonable

https://github.com/google-deepmind/mujoco/assets/140307654/3890126e-8663-4daa-9bc7-f40b4b1c9adc

Operating system: macOS Ventura
MuJoCo version: 3.0.0

saran-t, 2023-11-21:
@quagla looks like the SDF isn't taking scaling into account?

quagla, 2023-12-04:
Hi @DimaRubyAI sorry I missed this issue until now. The `scale` attribute is not taken into account by the SDF. For your case, I would suggest adding a new plugin instance
```
<instance name="torus_b">
  <config key="radius1" value="3.5"/>
  <config key="radius2" value="1.5"/>
</instance>
```
Would that help?

commit 623201ec (2023-12-08): Throw error if scale attribute is used with an SDF.

Fixes #1210.

PiperOrigin-RevId: 589101807
Change-Id: I3f2796de3438fbcdaf9221bddc5c9bd82aedb7cb
closed 2023-12-08 by quagla