google-deepmind/mujoco #1213: Mujoco 2.3.7, 3.0.1 not installable via wheel on MacOS

ikamensh, 2023-11-21:
Hello,

Many versions of mujoco could have been installed easily from a wheel, e.g. today I've tested 2.2.1, 2.3.2 versions.
However important latest versions don't find a wheel on pip:

```
$ pip install mujoco==2.3.2
Collecting mujoco==2.3.2
  Downloading mujoco-2.3.2-cp310-cp310-macosx_10_12_x86_64.**whl** (3.9 MB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 3.9/3.9 MB 4.5 MB/s eta 0:00:00
```

```
Collecting mujoco>=3.0.0 (from dm-control>=1.0.10->shimmy[dm-control,gym-v21]<2,>=1.2.1)
  Downloading mujoco-3.0.1.**tar.gz** (656 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━ 656.5/656.5 kB 5.0 MB/s eta 0:00:00
```

This leads to attempt to compile mujoco, which requires more elaborate setup, e.g. env variables with mujoco path. It would be helpful not to worry about it and get a wheel instead.

Environment:
MacOS 14.0, Python3.10

saran-t, 2023-11-21:
This should just work (I regularly install from wheel on my laptop).

Can you run `pip` in verbose mode to see why it thinks that the wheels on PyPI are not suitable for your machine?

nimrod-gileadi, 2023-12-04:
Could you give us the output of the following Python commands:

```python
import platfom
import sys

print('CPU:', platform.machine())
print('Mac Version:', platform.mac_ver())
print('Python version:', sys.version_info)
```

Could you try renaming the file to `mujoco-3.0.1-cp311-cp311-macosx_12_0_x86_64.whl` or `mujoco-3.0.1-cp311-cp311-macosx_10_12_x86_64.whl` and see what pip says?

ikamensh, 2023-12-05:
```
CPU: x86_64
Mac Version: ('10.16', ('', '', ''), 'x86_64')
Python version: sys.version_info(major=3, minor=11, micro=5, releaselevel='final', serial=0)
```

my os menu -> about this mac says: "macOS    14.1.1 (23B81)"

Renaming the wheel to 10_12 made it installable, and I could `import mujoco` with no error. Renaming the wheel to 12_0 produced pip error.
ERROR: mujoco-3.0.1-cp311-cp311-macosx_12_0_x86_64.whl is not a supported wheel on this platform.

nimrod-gileadi, 2023-12-05:
Hmm... so looks like `platform.mac_ver()` is reporting the wrong version, which is something that comes up with many Google search results. We might have to reverse our policy of targeting the last supported version of mac, if this is a common configuration.

To solve your immediate issue, I think you can set the environment variable `SYSTEM_VERSION_COMPAT=0`. I'll look into the consequences of releasing wheels that target 10_16 instead of 11_0.

ikamensh, 2023-12-05:
I've installed it using conda. I've installed conda with homebrew about 2 months ago, and used `conda create --name ct311 python=3.11` command.

nimrod-gileadi, 2023-12-05:
Does SYSTEM_VERSION_COMPAT=0 fix the issue for you?
Do you have it set to 1 on a fresh terminal?

ikamensh, 2023-12-06:
Hey, in a fresh terminal this variable is not set. Setting it to 0 makes pip resolve wheel `mujoco-3.0.1-cp311-cp311-macosx_11_0_x86_64.whl` and install it.

nimrod-gileadi, 2023-12-13:
We just pushed out MuJoCo 3.1.0, which has wheels targeting macosx_10_16. Does it successfully install on your system?

ikamensh, 2023-12-19:
yes, it installs on my system without any env variable magic. Nice! :)

ikamensh, 2023-12-19:
if you could please add a wheel that would automatically resolve for "mujoco<3", which currently is 2.3.7. I think before everyone migrates to mujoco3 this one will be often used.

nimrod-gileadi, 2023-12-19:
Done, for 2.3.7 and for 3.0.0. Please reopen if it doesn't work for you.

closed 2023-12-19 by nimrod-gileadi