google-deepmind/mujoco #1214: DMC `cartpole.xml` using `mjx` crashes

bayerj, 2023-11-21:
**How to ask for help**

Hi,

I'm a scholar and I'm trying to use MuJoCo for the cartpole on the gpu.

I'm looking for some help with the dmc cartpole xml file together with mjx, which can be found [here](https://github.com/google-deepmind/dm_control/blob/main/dm_control/mujoco/testing/assets/cartpole.xml).

Here is what I use for simulation, and which I would expect to work.

```
import jax.numpy as jnp
import mujoco as mj
from mujoco import mjx

model = mj.MjModel.from_xml_string(open("cartpole.xml").read())
mjx_model = mjx.device_put(model)
data = mjx.make_data(mjx_model)

data = data.replace(ctrl=jnp.ones(1))
mjx.step(mjx_model, data),
```

However, I get the following error. Is this mujoco file not supported anymore, or not yet supposed to work?

```
.venv/lib/python3.10/site-packages/mujoco/mjx/_src/device.py:108: UserWarning: Solver mjSOL_NEWTON is not supported, reverting to CG.
  warnings.warn(f'Solver {name} is not supported, reverting to CG.')
Traceback (most recent call last):
  File ".venv/lib/python3.10/site-packages/jax/_src/array.py", line 254, in __len__
    return self.shape[0]
IndexError: tuple index out of range

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "examples/policy-learning/meh.py", line 11, in <module>
    mjx.step(mjx_model, data),
  File ".venv/lib/python3.10/site-packages/mujoco/mjx/_src/forward.py", line 57, in wrapper
    res = fn(*args, **kwargs)
  File ".venv/lib/python3.10/site-packages/mujoco/mjx/_src/forward.py", line 347, in step
    d = forward(m, d)
  File ".venv/lib/python3.10/site-packages/mujoco/mjx/_src/forward.py", line 57, in wrapper
    res = fn(*args, **kwargs)
  File ".venv/lib/python3.10/site-packages/mujoco/mjx/_src/forward.py", line 329, in forward
    d = _actuation(m, d)
  File ".venv/lib/python3.10/site-packages/mujoco/mjx/_src/forward.py", line 57, in wrapper
    res = fn(*args, **kwargs)
  File ".venv/lib/python3.10/site-packages/mujoco/mjx/_src/forward.py", line 154, in _actuation
    force = scan.flat(
  File ".venv/lib/python3.10/site-packages/mujoco/mjx/_src/scan.py", line 192, in flat
    _check_input(m, args, in_types)
  File ".venv/lib/python3.10/site-packages/mujoco/mjx/_src/scan.py", line 131, in _check_input
    if len(a) != size[typ]:
  File ".venv/lib/python3.10/site-packages/jax/_src/array.py", line 256, in __len__
    raise TypeError("len() of unsized object") from err  # same as numpy error
TypeError: len() of unsized object
```

erikfrey, 2023-11-22:
Hi @bayerj - it looks like you are using MuJoCo 3.0.0 - can you instead try the [recently released 3.0.1](https://github.com/google-deepmind/mujoco/discussions/1215)?  I just tried cartpole.xml with it and it worked fine.  Thanks!

bayerj, 2023-11-22:
That worked! Thanks.

closed 2023-11-22 by bayerj