google-deepmind/mujoco #1219: Segfault when using passive viewer with flexcomp in the loaded xml

lbusellato, 2023-11-22:
Hi,

I'm a Master's student in Robotics and I'm trying to use MuJoCo w/ Python for simulating the contact of an UR5e robot's end-effector with a soft, deformable object.

To do so, I've used the scene provided in the Menagerie for the UR5, with minimal tweaks:

<details>
  <summary>Scene XML</summary>

```XML
<mujoco model="ur5e scene">
  <compiler angle="radian" meshdir="meshes/ur5e" autolimits="true"/>

  <option integrator="implicitfast"/>

  <option timestep="0.002">
    <flag warmstart="enable"></flag>
  </option>

  <option impratio="10" />

  <extension>
    <plugin plugin="mujoco.elasticity.solid"/>
  </extension>

  <statistic center="0.3 0 0.4" extent="0.8"/>

  <visual>
    <headlight diffuse="0.6 0.6 0.6" ambient="0.1 0.1 0.1" specular="0 0 0"/>
    <rgba haze="0.15 0.25 0.35 1"/>
    <global azimuth="120" elevation="-20"/>
  </visual>

  <asset>
    <texture type="skybox" builtin="gradient" rgb1="0.3 0.5 0.7" rgb2="0 0 0" width="512" height="3072"/>
    <texture type="2d" name="groundplane" builtin="checker" mark="edge" rgb1="0.2 0.3 0.4" rgb2="0.1 0.2 0.3"
      markrgb="0.8 0.8 0.8" width="300" height="300"/>
    <material name="groundplane" texture="groundplane" texuniform="true" texrepeat="5 5" reflectance="0.2"/>
  </asset>

  <default>
    <default class="ur5e">
      <material specular="0.5" shininess="0.25"/>
      <joint axis="0 1 0" range="-6.28319 6.28319" armature="0.1"/>
      <!-- position actuators -->
      <!-- <general gaintype="fixed" biastype="affine" ctrlrange="-6.2831 6.2831" gainprm="2000"
      biasprm="0 -2000 -400"
        forcerange="-150 150"/> -->
      <default class="size3">
        <default class="size3_limited">
          <joint range="-3.1415 3.1415" />
          <!-- <general ctrlrange="-3.1415 3.1415"/> -->
        </default>
      </default>
      <default class="size1">
        <general gainprm="500" biasprm="0 -500 -100" forcerange="-28 28"/>
      </default>
      <default class="visual">
        <geom type="mesh" contype="0" conaffinity="0" group="2"/>
      </default>
      <default class="collision">
        <geom type="capsule" group="3"/>
        <default class="eef_collision">
          <geom type="cylinder"/>
        </default>
      </default>
      <site size="0.001" rgba="0.5 0.5 0.5 0.3" group="4"/>
    </default>
  </default>

[... 177 lines cut ...]
  File "/usr/lib/python3.8/threading.py", line 932 in _bootstrap_inner
  File "/usr/lib/python3.8/threading.py", line 890 in _bootstrap

Thread 0x00007f1486f2b700 (most recent call first):
  File "/home/buse/mujoco-venv/lib/python3.8/site-packages/Xlib/protocol/display.py", line 561 in send_and_recv
  File "/home/buse/mujoco-venv/lib/python3.8/site-packages/Xlib/protocol/rq.py", line 1380 in reply
  File "/home/buse/mujoco-venv/lib/python3.8/site-packages/Xlib/protocol/rq.py", line 1368 in __init__
  File "/home/buse/mujoco-venv/lib/python3.8/site-packages/Xlib/ext/record.py", line 219 in __init__
  File "/home/buse/mujoco-venv/lib/python3.8/site-packages/Xlib/ext/record.py", line 238 in enable_context
  File "/home/buse/mujoco-venv/lib/python3.8/site-packages/pynput/_util/xorg.py", line 407 in _run
  File "/home/buse/mujoco-venv/lib/python3.8/site-packages/pynput/keyboard/_xorg.py", line 545 in _run
  File "/home/buse/mujoco-venv/lib/python3.8/site-packages/pynput/_util/__init__.py", line 210 in run
  File "/usr/lib/python3.8/threading.py", line 932 in _bootstrap_inner
  File "/usr/lib/python3.8/threading.py", line 890 in _bootstrap

Thread 0x00007f148d298740 (most recent call first):
  File "/usr/lib/python3.8/threading.py", line 302 in wait
  File "/usr/lib/python3.8/queue.py", line 170 in get
  File "/home/buse/mujoco-venv/lib/python3.8/site-packages/mujoco/viewer.py", line 473 in launch_passive
  File "/home/buse/Desktop/CoLeCT_sim/colect_sim/env/ur5_env.py", line 261 in __init__
  File "/home/buse/Desktop/CoLeCT_sim/main.py", line 10 in \<module>
Segmentation fault (core dumped)
</details>

This does not happen when removing the import of the flexcomp, in that case the viewer pops up and the simulation plays out as expected. I've tried looking at the documentation as well as other issues (#1193 had the same problem, but was closed with no solution), without luck. Can someone point me in the right direction? Thanks.

kevinzakka, 2023-11-22:
HI @lbusellato, does the model load in the native simulate window?

lbusellato, 2023-11-22:
Hello @kevinzakka, yes, I can use simulate to load, separately, either the ur5 or the deformable cube, but not with them together in the scene.

kevinzakka, 2023-11-22:
I lowered the timestep to `0.001` and it's much better behaved. Can you try that on your end?

lbusellato, 2023-11-23:
Tried that with simulate and it works. Oddly, simulate works fine also with the old timestep now, even though I did not change anything else (besides rebooting my system). 

In any case, viewer still segfaults. The window pops up for a split second, and I can see the left/right ui tabs, while the middle part is black.

giorgionicola, 2023-11-23:
Hi, I am the author of (https://github.com/google-deepmind/mujoco/issues/1193), I have downloaded the latest version of Mujoco 3.0.1 and fixed the segmentation error on all models except those using the plugin mujoco.elasticity.solid

I am reopening the issue

lbusellato, 2023-11-29:
Hello, thanks all for the quick response. Unfortunately I am still having issues. I tried installing mujoco from github head (with `pip install mujoco --pre -f https://py.mujoco.org/`, and it installs version 3.0.2.dev586380774 without issue. When I launch the viewer however, I get the following error:

```
ERROR: mj_stackAlloc: insufficient memory: max = 0, available = 0, requested = 24 (ne = 0, nf = 0, nefc = 0, ncon = 0)

Press Enter to exit ...
```

Am I missing something obvious here? Thanks all again for the help.

yuvaltassa, 2023-11-30:
Did you read [this](https://mujoco.readthedocs.io/en/latest/modeling.html#csize)?
TL;DR you want `<size memory="..."/>`.

That said, the `max = 0` bit in your printout is surprising, so maybe this is something else...

lbusellato, 2023-11-30:
Hello @yuvaltassa, tried with `<size memory="100"/>`, got the following:

```
ValueError: Error: engine error: mj_stackAlloc: insufficient memory: max = 100, available = 72, requested = 768 (ne = 0, nf = 0, nefc = 0, ncon = 0)
```

Tried increasing it further to `<size memory="10000"/>`, got the following:

```
ValueError: Error: engine error: mj_stackAlloc: insufficient memory: max = 10000, available = 760, requested = 9216 (ne = 0, nf = 0, nefc = 0, ncon = 0)
```

Since my system has 12GB of available RAM to work with, I wanted to increase it a lot more to see what would happen by setting `<size memory="4G"/>`, and I get this:

```
ERROR: mj_stackAlloc: insufficient memory: max = 0, available = 0, requested = 24 (ne = 0, nf = 0, nefc = 0, ncon = 0)

Press Enter to exit ...
```
Tried also the other suffixes (K and M), and none seems to set the max.

yuvaltassa, 2023-12-01:
Extremely strange!

1. If you avoid the K/M/G suffixes and just add zeros, it works?
2. This is with the exact models you posted above? Or did you tweak them since then? If yes I'll try to reproduce.
3. Please give the details of your system: OS, local languages, processor, whatever might be releveant.

commit 8c7c39a2 (2023-12-05): Lock the sim mutex in PhysicsLoop and PhysicsThread.

This prevents an invalid memory access when the stack is used in engine_vis.

Fixes #1219.

PiperOrigin-RevId: 588017796
Change-Id: I1ee18c5aa5e96ccaeebb0a467cd384355882b80b