google-deepmind/mujoco #1223: Accurate external forces at the moment of calling `mjcb_act_dyn`

VasilyRakche, 2023-11-23:
Hey MuJoCo team,

At the moment when `mjcb_act_dyn` is called from [mj_fwdActuation(m, d)](https://github.com/google-deepmind/mujoco/blob/f2025c6abf4980cbb7ad757484e406cb7868bca4/src/engine/engine_forward.c#L1021) all external forces are still not available (up to date), since further, in [mj_fwdConstraint(m, d)](https://github.com/google-deepmind/mujoco/blob/f2025c6abf4980cbb7ad757484e406cb7868bca4/src/engine/engine_forward.c#L1023) `qfrc_constraint` forces are recomputed and based on them `qacc` are updated.

As a result, one can not perceive all external forces at the moment when `mjcb_act_dyn` is called. My goal is to implement **custom dynamic** and for that its absolutely crucial to have this info. Can you recommend what can I do in order to get updated forces at the moment of calling `mjcb_act_dyn` ?

erez-tom, 2023-11-23:
Actuation is computed before contact forces because in many cases the actuation signal directly affects the forces in the system (and therefore affects the resulting contact forces).  When you have second-order actuators, the actuation input only changes some actuation state variables (called `act`), and only `act` affects the forces in the system.  In this specific case, you could choose to compute the actuation (which will change `act`) either before or after the physics step was computed (which integrates the effect of `act`'s value).

Does this description match your case?  Would you be content with the actuation signal at time t to have no effect on the state of the multibody system at time t (only on the actuator state, which will only affect the next step)?

VasilyRakche, 2023-11-25:
Hey, thanks for your answer. It took a bit longer to provide feedback since I wanted to write a bit of code (minimal example) to show what I mean. Its not straight forward, but I hope it can clarify details. 

First, regarding your comments. As I far as I see, you mean implementing my own **ode** which can be handled through `act_dot` being integrated? Thats kinda what I am doing already, but the problem is that I need to get an accurate **force feedback** from the environment in order to represent the actuator dynamics properly. Within my custom dynamics, I also update the `qfrc_applied` within `mjcb_act_dyn`.

What I try to represent in a system is a clutch which would block the movement of the actuator:
```
- Clutch engaged:
      - qvel = 0
      - qacc = 0
- Clutch disengaged
      - no constraints on qvel 
      - no constraints on qacc 
```
The way I am trying to do is:
- Having a `Cp` as a selector matrix (indicating which `qvel` should be `0`) one can follow derivation from [this paper](https://ieeexplore.ieee.org/abstract/document/9981928) [1] in order to get the forces **($\lambda$)** that should be applied in order to compensate all external forces:

> [image]
> [image]

This successfully works in [this code repo](https://github.com/VasilyRakche/pendulum_force_compensation). (One can notice that the printed `joint_0` position is not changing) However, as soon as I add constraints to the [`joint_1`](https://github.com/VasilyRakche/pendulum_force_compensation/blob/b2b4b9a2f4c1dc22c0b325a1f71d1b72cafe29d7/tripplependulum.xml#L23) as follows:
```
				<joint name="joint_1" range="-3.14 3.14" axis = "1 0 0" pos="0 0 -0.2"/>
```

You will notice that from when `qfrc_constraint` becomes non-zero for any of the actuators (that's valid due to coupling), I can not accurately estimate and compensate the forces. Thus `joint_0` will start moving (check the printed messages in the terminal). I am not sure how it can be handled by delaying the input with `act`.

Minimal example provided [here](https://github.com/VasilyRakche/pendulum_force_compensation/tree/master)

_[1] D. Ossadnik et al., "BSA - Bi-Stiffness Actuation for optimally exploiting intrinsic compliance and inertial coupling effects in elastic joint robots," 2022 IEEE/RSJ International Conference on Intelligent Robots and Systems (IROS), Kyoto, Japan, 2022, pp. 3536-3543, doi: 10.1109/IROS47612.2022.9981928._

closed 2023-11-26 by yuvaltassa