google-deepmind/mujoco #1226: Combining Composites

tudorjnu, 2023-11-24:
I want to connect two cables in order to model different stiffness sections. The model below is an attempt at this, but I am not sure if I am on the right track. They seem to almost be connected, but when I extended it to more bodies, it is obvious it is not the correct approach.

Do you have any recommendations on how to do this effectively?

<details>
  <summary>minimal XML</summary>

```XML
<mujoco model="composite">
<extension>
    <plugin plugin="mujoco.elasticity.cable"/>
</extension>
<size memory="1G"/>
<worldbody>
    <body name="my_composite">
        <composite prefix="A" type="cable" curve="s" count="5" size="0.1" initial="none">
            <geom type="capsule" size="0.0008" rgba="1 1 1 1" condim="1"/>
            <joint kind="main" damping="0.001" stiffness="0.001" margin="0.02"/>
            <plugin plugin="mujoco.elasticity.cable">
                <config key="bend" value="1e9"/>
                <config key="flat" value="true"/>
                <config key="vmax" value="0.005"/>
            </plugin>
        </composite>
        <composite prefix="B" type="cable" curve="s" count="5" size="0.1" offset="0.1 0 0" initial="ball">
            <geom type="capsule" size="0.0008" rgba="0 1 0 1" condim="1"/>
            <joint kind="main" damping="0.001" stiffness="0.001" margin="0.02"/>
            <plugin plugin="mujoco.elasticity.cable">
                <config key="bend" value="1e9"/>
                <config key="flat" value="true"/>
                <config key="vmax" value="0.005"/>
            </plugin>
        </composite>
    </body>
</worldbody>
<equality>
    <connect name="right_boundary" body1="AB_last" body2="BB_first" anchor="0 0 0" />
</equality>
<contact>
    <exclude body1="AB_last" body2="BB_first"/>
</contact>
</mujoco>

```

</details>

<details>
  <summary>More Bodies</summary>

```XML
<mujoco model="composite">
<extension>
    <plugin plugin="mujoco.elasticity.cable"/>
</extension>
<size memory="1G"/>
<worldbody>
    <body name="my_composite">
        <composite prefix="A" type="cable" curve="s" count="50" size="0.3" initial="none">
            <geom type="capsule" size="0.0008" rgba="1 1 1 1" condim="1"/>
            <joint kind="main" damping="0.001" stiffness="0.001" margin="0.02"/>
            <plugin plugin="mujoco.elasticity.cable">
                <config key="bend" value="1e9"/>
                <config key="flat" value="true"/>
                <config key="vmax" value="0.005"/>
            </plugin>
        </composite>
        <composite prefix="B" type="cable" curve="s" count="10" size="0.1" offset="0.3 0 0" initial="ball">
            <geom type="capsule" size="0.0008" rgba="0 1 0 1" condim="1"/>
            <joint kind="main" damping="0.001" stiffness="0.001" margin="0.02"/>
            <plugin plugin="mujoco.elasticity.cable">
                <config key="bend" value="1e9"/>
                <config key="flat" value="true"/>
                <config key="vmax" value="0.005"/>
            </plugin>
        </composite>
    </body>
</worldbody>
<equality>
    <weld name="right_boundary" body1="AB_last" body2="BB_first" anchor="0 0 0" />
</equality>
<contact>
    <exclude body1="AB_last" body2="BB_first"/>
</contact>
</mujoco>
```

</details>

_Note: I tried including this in my previous issue but seems like I could not reopen the issue (https://github.com/google-deepmind/mujoco/issues/1195)_

Thank you!

quagla, 2023-11-27:
Hi, I don't see a way of doing that at the moment unfortunately. An option would be to move the plugin attributes from the plugin instance to the bodies, so you could load the composite, save the xml, and then edit each body stiffness by hand (the cable is formed by a series of bodies e.g. capsules). @yuvaltassa maybe it's time to move the cable plugin to the engine?

tudorjnu, 2023-11-28:
I see. Given the internal representation (using the `plugin` instance), I could write an `mjcf` script to create this, but I thought there might be a simpler, more elegant way. 

Thanks for the response! :smile:

closed 2024-01-02 by yuvaltassa

tudorjnu, 2024-07-10:
Hello @quagla ! Do you reckon there is a better way to do this at this point? Maybe using the `replicate` option? What do you think?