google-deepmind/mujoco #1227: MJX does not keep equality constraints

EnricoMingo, 2023-11-24:
Hi,

I'm trying to use MuJoCo (MJX) to simulate a humanoid bipedal robot with several closed linkages. I am using MJX inside BRAX and when I visualize some seconds of trajectory I clearly see that the equality constraints that I put inside the model are not kept. These constraints normally work using "normal" MUJOCO.

I am using the default settings for MjModel:

    mj_model.opt.solver = mujoco.mjtSolver.mjSOL_NEWTON 
    mj_model.opt.iterations = 100
    mj_model.opt.ls_iterations = 50

notice that I have 32 equality constraints in my model (for most of them solref = "0.005 1").
Thank you for your help!

yuvaltassa, 2023-11-24:
Can you please attach a minimal model that reproduces this issue?

EnricoMingo, 2023-11-24:
Thank you for the prompt reply. I am following the tutorial [here](https://colab.research.google.com/github/google-deepmind/mujoco/blob/main/mjx/tutorial.ipynb) and I changed the model from the humanoid to my robot. I also have commented out the following lines of code:

`mj_model.opt.solver = mujoco.mjtSolver.mjSOL_CG`
`mj_model.opt.iterations = 6`
`mj_model.opt.ls_iterations = 6`

I check the output from the "# grab a trajectory" part of the tutorial. I thought it was related with the different settings of the solver so I tried to put the default values as well (the ones in the previous post), but I am still having this issue.
The model of the robot is quite complex, I will try to prepare a smaller version with less constraints and see if the error remains.

sck-robot, 2023-11-25:
I seem to be having a similar issue. I created a closed-loop four-link model, which works properly under mujoco. I used `connect` to implement this closed loop. But under mjx, the closed loop linkage doesn't seem to be working properly. I have provided a code to reproduce this problem. Does anyone have any ideas about this?
Thanks a lot.

Uploading loop_in_mucojo.mp4…

https://github.com/google-deepmind/mujoco/assets/60764118/b312a9dd-2e06-4236-9bf4-647bc8c22c49

`import mujoco
import jax
import mujoco_viewer
import time
from mujoco import mjx

# Create a closed-loop linkage
XML=r"""
<mujoco model="my_bipedal">

  <compiler eulerseq="xyz" angle="radian" meshdir="assets" texturedir="assets" autolimits="true"/>

  <option timestep="0.001"/>

  <default class="pantograph">
      <equality solimp="0.005 1"/>
  </default>

  <worldbody>

    <body name="Link0" pos="0 0 0.5" euler="0 0 0">

        <inertial pos="0 0 0" mass="1"   />    
        <geom type="capsule" size = "0.01" fromto = "-0.25 0 0 0.25 0 0" rgba="1 0.1 0.1 1"/>
        <body name="Link1" pos ="-0.25 0 0">
            <geom type="capsule" size = "0.01" fromto = "0 0 0 0 0 -0.2" rgba="1 0.1 0.1 1"/>
            <inertial pos="0.03 0 -0.02" mass="1"   diaginertia = "1e-3 1e-3 1e-3"/>    
            <joint name="joint1" pos = "0 0 0" axis="0 1 0"/>
        </body>

        <body name="Link2" pos ="0.25 0 0">
            <geom  type="capsule" size = "0.01" fromto = "0 0 0 0 0 -0.2" rgba="1 0.1 0.1 1"/>
            <inertial pos="0 0 -0.02" mass="1"   diaginertia = "1e-3 1e-3 1e-3"/>    
            <joint name="joint2" pos = "0 0 0" axis="0 1 0"/>

            <body name="Link3" pos ="0 0 -0.2" euler="0 0 0">
                <geom type="capsule" size = "0.01" fromto = "0 0 0 -0.5 0 0" rgba="1 0.1 0.1 1"/>
                <inertial pos="-0.25 0 0" mass="1"  diaginertia = "1e-3 1e-3 1e-3"/>    
                <joint name="joint3" pos = "0 0 0" axis="0 1 0"/>
            </body>

        </body>

    </body>

  </worldbody>

  <equality>
      <connect class="pantograph" name="Loop" body1="Link1" body2="Link3" anchor="0 0 -0.2"/>
  </equality>

</mujoco>

"""

# Choose mujoco or mjx for simulation
# mode = 'mujoco'             
mode = 'mjx'                        

mjc_model =mujoco.MjModel.from_xml_string(XML)
mjc_data = mujoco.MjData(mjc_model)
mjx_model = mjx.device_put(mjc_model)
mjx_data = mjx.device_put(mjc_data)
viewer = mujoco_viewer.MujocoViewer(mjc_model, mjc_data, width=1800,height=1200)

@jax.jit
def mjx_step_forward(mjx_model,mjx_data):
    def f(data, _):
        return (
            mjx.step(mjx_model, data),
            None,
        )
    mjx_data, _ = jax.lax.scan(f, mjx_data, (), 3)
    return mjx_data

if mode=='mjx':
    mjx_data = mjx.forward(mjx_model, mjx_data)

count = 0
while True:

    # the loop is closed in the simulation of mujoco
    if mode =='mujoco':
        mujoco.mj_step(mjc_model, mjc_data, nstep=3)

    # the loop is opened in the simulation of mjx
    if mode =='mjx':
        mjx_data = mjx_step_forward(mjx_model,mjx_data)
        mjc_data.qpos = jax.device_get(mjx_data.qpos).copy()
        mjc_data.qvel = jax.device_get(mjx_data.qvel).copy()
        mujoco.mj_forward(mjc_model, mjc_data)
    
    count +=1

    if count==3:
        viewer.render()
        time.sleep(0.001 * count)
        count = 0

`

EnricoMingo, 2023-11-25:
@sck-robot thank you for your example. I was planning to do something similar next week but you anticipated me :)

sck-robot, 2023-11-25:
您的信件已收到，我会尽快处理！

EnricoMingo, 2023-11-25:
🙄

heleiduan, 2023-11-30:
echoing a similar closed-loop violations seen in an existing model from [mujoco_menagerie](https://github.com/google-deepmind/mujoco_menagerie). an earlier thread #1129 includes the example code that we tried as well.

erikfrey, 2023-11-30:
Hi all - confirming that I can repro this on my end using the model provided by @sck-robot.  Will report back soon with a fix.

erikfrey, 2023-11-30:
OK, this is fixed as of 67fa7c1dc21b535b9b5f906830f5ebf81dbf4d7e which will be available in MuJoCo 3.0.2.  In the meantime, I can explain the bug if you would like to fix it in your user code and still use 3.0.1:

MJX is incorrectly setting `d.ne` to 0 `mjx.device_put` and is then ignoring equality constraints during the step.  For the time being, you can fix this by using `mjx.make_data` to initialize `mjx.Data` instead of `mjx.device_put`.  So in your example @sck-robot make the following change:

```diff
79c79
<     mjx_data = mjx.forward(mjx_model, mjx_data)
---
>     mjx_data = mjx.make_data(mjx_model)
```

In MuJoCo 3.0.2 we will offer some new functions for placing data on device: `mjx.put_data` and `mjx.put_model` that work better in these scenarios.

sck-robot, 2023-12-01:
Thank you @erikfrey  for solving the issue so instantly! Now I can create the closed loop correctly under MJX 3.0.1 with your advice. I'm really looking forward to MJX's new features. Thank you again for your efforts.

EnricoMingo, 2023-12-01:
I was able to fix my script as well. It is based on the mujoco_mjx environment for Brax example. It was just needed to apply the suggestion of @erikfrey inside the `pipeline_init` function.

heleiduan, 2023-12-01:
@erikfrey Thank you for taking care of this! The closed loop in robot model is able to work properly with compiled version of commit https://github.com/google-deepmind/mujoco/commit/67fa7c1dc21b535b9b5f906830f5ebf81dbf4d7e.