google-deepmind/mujoco #1229: Simulation Stability / Tuning Actuators

peterdavidfagan, 2023-11-25:
Hi,

I'm a PhD student and I'm trying to use MuJoCo for simulating [object rearrangement tasks](https://github.com/peterdavidfagan/rearrangement_benchmark). I am using these tasks to evaluate different approaches to learning policies for robot perception and control.

I'm looking for some help with creating stable simulations using MuJoCo. In particular, I wish to ensure that actuators on robot arms (Franka Emika Panda, UFactory Lite6) run reliably, for instance if I pass a feasible joint position target to a position servo I expect the controller to achieve this target while respecting control limits of the robot. I also wish to ensure that gripper models work reliably when grasping objects.

Here is an explanation of my question:

I am attempting basic pick and place behaviours in an object manipulation benchmark that leverages MuJoCo. I am struggling to run reliable control on existing models. Thus far I have attempted to configure physics time stepping ([here](https://github.com/peterdavidfagan/rearrangement_benchmark/blob/a9d59e905cc53cb8e33a3891eb4a1b8441de2d5b/rearrangement_benchmark/env_components/base_scene.py#L27)) & control time stepping ([here](https://github.com/peterdavidfagan/rearrangement_benchmark/blob/a9d59e905cc53cb8e33a3891eb4a1b8441de2d5b/rearrangement_benchmark/task.py#L42)) as well as the actuator settings for the above robots. I am directly controlling the robots by passing joint targets to robot actuator models, I have mostly tested with direct position controllers but plan to attempt using PD controllers.

I wished to ask what is the best practice in this situation. Should I fix my sim step and control step and then tune controllers to this? If I am tuning controllers is it expected to perform this tuning manually or does there exist and open-source package for MuJoCo that provides automated tuning. 

My current set of code can be found [here](https://github.com/peterdavidfagan/rearrangement_benchmark). To reproduce what I am observing the following is required:

- System installation of poetry
- Clone repo and submodules (Note: the following submodules can be removed `vcp`, `ros2_robotics_research_toolkit`, `transporter_networks`)
- Install poetry env with `poetry install`
- Run the following script `python transporter_network_data_generation.py` 

Here is a screenshot / video, illustrating my question:
[Screencast from 11-25-2023 02:23:43 PM.webm](https://github.com/google-deepmind/mujoco/assets/42982057/60c7f84f-aa36-4a50-832c-a5a44bcf450d)

peterdavidfagan, 2023-11-25:
Update: I believe these issues relate to not constraining the velocity of motions, interacting with the robot model via the interactive viewer demonstrates that most motions are stable when I smoothly alter the position target.

kevinzakka, 2023-11-25:
How small are you making the timestep? In my experience, [0.005](https://github.com/peterdavidfagan/rearrangement_benchmark/blob/main/rearrangement_benchmark/env_components/base_scene.py#L27) is too high for the Robotiq.

peterdavidfagan, 2023-11-25:
Thanks @kevinzakka, 

> In my experience, [0.005](https://github.com/peterdavidfagan/rearrangement_benchmark/blob/main/rearrangement_benchmark/env_components/base_scene.py#L27) is too high for the Robotiq.

I'll ensure to keep it below this value, since earlier today I believe the last setting I worked with was a physics timestep of `0.001` and control timestep to `0.005`. The simulation seemed more stable with these settings (but still getting joint velocities that were quite high). 

> Unrelated by why is your gravity -1?

I think this is due to the values in the tutorials, when I started using these libraries I was referencing the Colabs for examples. In the following [Colab](https://colab.research.google.com/github/deepmind/dm_robotics/blob/main/py/moma/moma_tutorial.ipynb) the value `-1` is used rather than `-9.8`, I will update this and rerun my simulation.

**I plan to clean my code tomorrow morning as in its current state it is somewhat unclean. My next planned step was to add a velocity actuator to my robot config and enforce the ctrl limit constraint in order to address the joint velocity issue as it seems that the velocities are currently too large.

closed 2023-12-04 by yuvaltassa

peterdavidfagan, 2023-12-04:
Ok understood, thanks for  the feedback on this. For now I'll replace the robotiq gripper with the default for the Franka Emika Panda. 

> Not clear to me what's the benefit of an external PD controller as MuJoCo already does this and more stably than you can (the damping is integrated implicitly if you use implicit or implicitfast), but suit yourself.

This is good to know, I am honestly still learning this information is useful. I haven't tried combining more than one actuator on a joint in mujoco to get PD controllers but I will look into this.

> The Robotiq gripper is not well modeled at the moment, people have been complaining about it for while: https://github.com/google-deepmind/mujoco/issues/906 https://github.com/google-deepmind/mujoco/issues/996

Thanks for this information it does save me some time, I was only starting to look into the model parameters.

peterdavidfagan, 2023-12-04:
For the purposes of others who may also be debugging and or learning. The gripper qualitative behaviour demonstrated in the previous image only occurs for the custom PD controller with inverse dynamics.

[Screencast from 12-04-2023 03:55:36 PM.webm](https://github.com/google-deepmind/mujoco/assets/42982057/9a3f6c87-d4bc-4dde-8a7b-3858dbb28981)

kevinzakka, 2023-12-04:
Hi @peterdavidfagan, I've been using the Robotiq with a Panda arm and an operational space controller and haven't had any issues with the gripper. Do you want to compare notes with simulation settings and I can maybe try to help out?

peterdavidfagan, 2023-12-04:
Hi @kevinzakka, this would be super helpful but only if you have the time and it doesn't distract from your current research/tasks. 

I'd be happy to give a full outline here of my current settings with clean reproducible examples. Does it suit if I post these early tomorrow as I am finishing CMA-ES this evening? Also I feel this will be really helpful for others looking to use MuJoCo especially with the Franka Emika + Robotiq combo. 

Also thanks for all the work you did in open sourcing existing models, these models are really valuable.

kevinzakka, 2023-12-04:
@peterdavidfagan Sent you a DM on twitter, feel free to followup there and schedule a video call.

peterdavidfagan, 2023-12-06:
@kevinzakka's OSC controller implementation as a reference works for grasping so far (more tests to be completed). 

The implementation can be found [here](https://github.com/kevinzakka/mjctrl/blob/main/opspace.py), many thanks for sharing this.

peterdavidfagan, 2023-12-28:
Further follow-up in case anyone is reading this retrospectively. I implemented OSC and was able to perform manual clipping such that the simulation behaviour was reasonable [here](https://github.com/peterdavidfagan/mujoco_controllers/blob/main/mujoco_controllers/osc.py).

This being said it is possible to explicitly limit the end effector velocity in OSC with the formulation outlined [here](https://studywolf.wordpress.com/2016/11/07/velocity-limiting-in-operational-space-control/). I have not implemented this yet but will likely update the OSC implementation to be of this form. 

In order to directly constrain joint velocities one could use differential inverse kinematics with the appropriate constraints.

Tadinu, 2024-09-27:
@peterdavidfagan For velocity limiting, here you may find the implementation by the blog author
https://github.com/abr/abr_control/blob/main/abr_control/controllers/osc.py

peterdavidfagan, 2024-09-27:
Thanks for sharing the above @Tadinu.