google-deepmind/mujoco #1232: CMake error when installing mujoco in conda environment on MacOS

maxholsman, 2023-11-26:
Hi everyone,

I am trying to install mujoco on my Mac for a research project and keep getting the following error when running 'pip install mujoco' or 'pip install 'gymnasium[mujoco]'. The full error is copied after this post.

subprocess.CalledProcessError: Command '['cmake', '--build', '.', '-j16', '--config', 'Release']' returned non-zero exit status 2.

Interestingly, when I run pip install mujoco outside of a conda environment, I don't seem to get any errors, however I am still unable to run python scripts that use mujoco directly or through gymnasium. Any idea on how to solve this issue?

(the entire output is obviously much longer - I have only copied the last lines close to the error)

  [  6%] Building CXX object _deps/abseil-cpp-build/absl/time/CMakeFiles/absl_time_zone.dir/internal/cctz/src/time_zone_if.cc.o
  [  6%] Building CXX object _deps/abseil-cpp-build/absl/time/CMakeFiles/absl_time_zone.dir/internal/cctz/src/time_zone_impl.cc.o
  [  6%] Building CXX object _deps/abseil-cpp-build/absl/time/CMakeFiles/absl_time_zone.dir/internal/cctz/src/time_zone_info.cc.o
  [  6%] Building CXX object _deps/abseil-cpp-build/absl/time/CMakeFiles/absl_time_zone.dir/internal/cctz/src/time_zone_lookup.cc.o
  [  6%] Building CXX object _deps/abseil-cpp-build/absl/time/CMakeFiles/absl_time_zone.dir/internal/cctz/src/time_zone_posix.cc.o
  [ 13%] Building CXX object _deps/abseil-cpp-build/absl/time/CMakeFiles/absl_time_zone.dir/internal/cctz/src/time_zone_libc.cc.o
  cc1plus: error: '-Werror=partial-availability': no option '-Wpartial-availability'
  cc1plus: error: '-Werror=partial-availability': no option '-Wpartial-availability'
  cc1plus: error: '-Werror=partial-availability': no option '-Wpartial-availability'
  cc1plus: error: '-Werror=partial-availability': no option '-Wpartial-availability'
  cc1plus: error: '-Werror=partial-availability': no option '-Wpartial-availability'
  cc1plus: error: '-Werror=partial-availability': no option '-Wpartial-availability'
  cc1plus: error: '-Werror=unguarded-availability': no option '-Wunguarded-availability'
  cc1plus: error: '-fno-fat-lto-objects' are supported only with linker plugin
  cc1plus: error: '-Werror=unguarded-availability': no option '-Wunguarded-availability'
  cc1plus: error: '-Werror=unguarded-availability': no option '-Wunguarded-availability'
  cc1plus: error: '-fno-fat-lto-objects' are supported only with linker plugin
  cc1plus: error: '-fno-fat-lto-objects' are supported only with linker plugin
  cc1plus: error: '-Werror=unguarded-availability': no option '-Wunguarded-availability'
  cc1plus: error: '-fno-fat-lto-objects' are supported only with linker plugin
  make[2]: *** [_deps/abseil-cpp-build/absl/time/CMakeFiles/absl_time_zone.dir/internal/cctz/src/time_zone_impl.cc.o] Error 1
  cc1plus: error: '-Werror=unguarded-availability': no option '-Wunguarded-availability'
  cc1plus: error: '-Werror=unguarded-availability': no option '-Wunguarded-availability'
  cc1plus: error: '-fno-fat-lto-objects' are supported only with linker plugin
  make[2]: *** [_deps/abseil-cpp-build/absl/time/CMakeFiles/absl_time_zone.dir/internal/cctz/src/time_zone_libc.cc.o] Error 1
  make[2]: *** [_deps/abseil-cpp-build/absl/time/CMakeFiles/absl_time_zone.dir/internal/cctz/src/time_zone_if.cc.o] Error 1
  make[2]: *** [_deps/abseil-cpp-build/absl/time/CMakeFiles/absl_time_zone.dir/internal/cctz/src/time_zone_lookup.cc.o] Error 1
  cc1plus: error: '-fno-fat-lto-objects' are supported only with linker plugin
  make[2]: *** [_deps/abseil-cpp-build/absl/time/CMakeFiles/absl_time_zone.dir/internal/cctz/src/time_zone_info.cc.o] Error 1
  make[2]: *** [_deps/abseil-cpp-build/absl/time/CMakeFiles/absl_time_zone.dir/internal/cctz/src/time_zone_posix.cc.o] Error 1
  make[1]: *** [_deps/abseil-cpp-build/absl/time/CMakeFiles/absl_time_zone.dir/all] Error 2
  make: *** [all] Error 2
  Configuring CMake with the following arguments:
      -DPython3_ROOT_DIR:PATH=/Users/maximilianholsman/anaconda3/envs/mujoco_env
      -DPython3_EXECUTABLE:STRING=/Users/maximilianholsman/anaconda3/envs/mujoco_env/bin/python
      -DCMAKE_MODULE_PATH:PATH=/private/var/folders/sm/k5l3gtmn1cq_8dmrhdnxzytw0000gn/T/pip-install-ie04o9gh/mujoco_b298c8efc5e94ee7bd98f43646b9dbc2/mujoco/cmake
      -DCMAKE_BUILD_TYPE:STRING=Release
      -DCMAKE_LIBRARY_OUTPUT_DIRECTORY:PATH=build/temp.macosx-10.9-x86_64-cpython-39
      -DCMAKE_INTERPROCEDURAL_OPTIMIZATION:BOOL=ON
      -DCMAKE_Fortran_COMPILER:STRING=
      -DBUILD_TESTING:BOOL=OFF
      -DMUJOCO_FRAMEWORK_DIR:PATH=/Applications/mujoco/MuJoCo.app/Contents/Frameworks
      -DPython3_LIBRARY=/Users/maximilianholsman/anaconda3/envs/mujoco_env/lib/python3.9
      -DPython3_INCLUDE_DIR=/Users/maximilianholsman/anaconda3/envs/mujoco_env/include/python3.9
  Building all extensions with CMake
  Traceback (most recent call last):
    File "/Users/maximilianholsman/anaconda3/envs/mujoco_env/lib/python3.9/site-packages/pip/_vendor/pyproject_hooks/_in_process/_in_process.py", line 353, in <module>
      main()
    File "/Users/maximilianholsman/anaconda3/envs/mujoco_env/lib/python3.9/site-packages/pip/_vendor/pyproject_hooks/_in_process/_in_process.py", line 335, in main
      json_out['return_val'] = hook(**hook_input['kwargs'])
    File "/Users/maximilianholsman/anaconda3/envs/mujoco_env/lib/python3.9/site-packages/pip/_vendor/pyproject_hooks/_in_process/_in_process.py", line 251, in build_wheel
      return _build_backend().build_wheel(wheel_directory, config_settings,
    File "/private/var/folders/sm/k5l3gtmn1cq_8dmrhdnxzytw0000gn/T/pip-build-env-stgql3ll/overlay/lib/python3.9/site-packages/setuptools/build_meta.py", line 404, in build_wheel
      return self._build_with_temp_dir(
    File "/private/var/folders/sm/k5l3gtmn1cq_8dmrhdnxzytw0000gn/T/pip-build-env-stgql3ll/overlay/lib/python3.9/site-packages/setuptools/build_meta.py", line 389, in _build_with_temp_dir
      self.run_setup()
    File "/private/var/folders/sm/k5l3gtmn1cq_8dmrhdnxzytw0000gn/T/pip-build-env-stgql3ll/overlay/lib/python3.9/site-packages/setuptools/build_meta.py", line 311, in run_setup
      exec(code, locals())
    File "<string>", line 334, in <module>
    File "/private/var/folders/sm/k5l3gtmn1cq_8dmrhdnxzytw0000gn/T/pip-build-env-stgql3ll/overlay/lib/python3.9/site-packages/setuptools/__init__.py", line 103, in setup
[... 12 lines cut ...]
      self.run_command("build")
    File "/private/var/folders/sm/k5l3gtmn1cq_8dmrhdnxzytw0000gn/T/pip-build-env-stgql3ll/overlay/lib/python3.9/site-packages/setuptools/_distutils/cmd.py", line 318, in run_command
      self.distribution.run_command(command)
    File "/private/var/folders/sm/k5l3gtmn1cq_8dmrhdnxzytw0000gn/T/pip-build-env-stgql3ll/overlay/lib/python3.9/site-packages/setuptools/dist.py", line 963, in run_command
      super().run_command(command)
    File "/private/var/folders/sm/k5l3gtmn1cq_8dmrhdnxzytw0000gn/T/pip-build-env-stgql3ll/overlay/lib/python3.9/site-packages/setuptools/_distutils/dist.py", line 988, in run_command
      cmd_obj.run()
    File "/private/var/folders/sm/k5l3gtmn1cq_8dmrhdnxzytw0000gn/T/pip-build-env-stgql3ll/overlay/lib/python3.9/site-packages/setuptools/_distutils/command/build.py", line 131, in run
      self.run_command(cmd_name)
    File "/private/var/folders/sm/k5l3gtmn1cq_8dmrhdnxzytw0000gn/T/pip-build-env-stgql3ll/overlay/lib/python3.9/site-packages/setuptools/_distutils/cmd.py", line 318, in run_command
      self.distribution.run_command(command)
    File "/private/var/folders/sm/k5l3gtmn1cq_8dmrhdnxzytw0000gn/T/pip-build-env-stgql3ll/overlay/lib/python3.9/site-packages/setuptools/dist.py", line 963, in run_command
      super().run_command(command)
    File "/private/var/folders/sm/k5l3gtmn1cq_8dmrhdnxzytw0000gn/T/pip-build-env-stgql3ll/overlay/lib/python3.9/site-packages/setuptools/_distutils/dist.py", line 988, in run_command
      cmd_obj.run()
    File "<string>", line 153, in run
    File "<string>", line 292, in _configure_cmake
    File "/Users/maximilianholsman/anaconda3/envs/mujoco_env/lib/python3.9/subprocess.py", line 373, in check_call
      raise CalledProcessError(retcode, cmd)
  subprocess.CalledProcessError: Command '['cmake', '--build', '.', '-j16', '--config', 'Release']' returned non-zero exit status 2.
  [end of output]
note: This error originates from a subprocess, and is likely not a problem with pip.
ERROR: Failed building wheel for mujoco
Failed to build mujoco
ERROR: Could not build wheels for mujoco, which is required to install pyproject.toml-based projects

molysgaard, 2023-11-26:
The problem is that your conda environment has a different `g++`.
And when it is given the warning arguments during CMake configuration, it errors that it doesn't support it.

Eg these lines:
```
 cc1plus: error: '-Werror=unguarded-availability': no option '-Wunguarded-availability'
```

To debug this, you can do two things:

1. Check the log to see exactly which compiler CMake chooses both outside and inside the conda environment. They will be different.
2. Let's say CMake reports that it chooses `g++` both inside and outside. Then you run `g++ --version` both inside and outside the environment, to figure out which versions both are and probably they will be different.

maxholsman, 2023-11-28:
Thank you! When trying to run installation outside of the environment to see difference in compiler used, I realized I was installing the prebuilt binary outside of the environment, whereas I was building from the source inside the environment, which is what was causing the issue. All sorted now!

closed 2023-12-03 by kevinzakka