google-deepmind/mujoco #1235: building a kinematic tree with flexcomp

yswi, 2023-11-27:
I hope to attach a flexible body to a rigid object.

When I compose a kinematic tree with FlexComp, it seems that the components are not rigidly attached to the parent. For instance, in the XML below, I observe the soft body falling due to gravity. (If I turn off gravity, the object stays in place.)

I wonder if this is a potential bug.
Thank you.

```
    <body name="box" pos=".1 0 1.5">
      <joint name="box" type="slide" />
      <geom type="box" size=".05 .05 .05"/>

      <flexcomp name="softbody" type="grid" count="8 4 20" spacing=".01 .01 .01" pos="0. 0. -0.15"
              radius=".0" rgba="0 .7 .7 1"  dim="3" mass="2">
        <contact condim="3" solref="0.01 1" solimp=".95 .99 .0001" selfcollide="none"/>
        <plugin plugin="mujoco.elasticity.solid">
          <config key="poisson" value="0.2"/>
          <!--Units are in Pa (SI)-->
          <config key="young" value="5e5"/>
          <config key="damping" value="0.002"/>
      </plugin>
      </flexcomp>
    </body>
```

[image]

quagla, 2023-11-27:
Hi, yes you found an issue that prevents doing that with plugins. I have a fix that will likely be pushed this week and will be part of the next release. In the meantime, I can suggest you to use `<edge equality="true"/>` in flexcomp and remove the plugin. See also https://mujoco.readthedocs.io/en/latest/XMLreference.html#flexcomp-edge. Or alternatively, you can pull the flex out of the body and add constraints (but they will be soft).

quagla, 2023-11-27:
I should also add that in any case you have to add a `pin` to your flex (https://mujoco.readthedocs.io/en/latest/XMLreference.html#flexcomp-pin) since by default all flex vertices have 3 dofs, i.e. they are not automatically anchored to the parent. It's the pin that has issues with the plugins and that will be fixed by my update.