google-deepmind/mujoco #1237: Add my own sdf to MuJoCo simulator and python code

DimaRubyAI, 2023-11-28:
Hi,

I'm a CV/ML engineer and I'm trying to use MuJoCo for Physical simulations.
I'm looking for some help with creating my own sdf.
As a first step I simply took an existing object from plugin sdf (torus) and rename it as disk (later on I want to change the properties of it).
As I understand I should build the project in order to use a my own sdf, am I wrong?
I built from source by using [this](https://mujoco.readthedocs.io/en/latest/programming/#building-mujoco-from-source) tutorial.
What happens, that when I run the built simulator from the bin folder even the pre-built sdfs like torus don't found by it.
[image]
What am i doing wrong? or basically what is the step that should be done in order to use my own sdf in simulator and python code?

DimaRubyAI, 2023-11-28:
@yuvaltassa, 
Sorry, I forgot to mention, I updated the CMakeLists.txt file as well.
But again the problem is not there, even "torus" (the one that included in the original source) not working for me.
When I try to drag&drop any xml model which include torus or any other sdf from the original examples to the simulator that I built I got the error.
```
XML Error: unknown plugin 'mujoco.sdf.torus'
Element 'plugin', line 4
```
should I build the sdf separately? copy some object/header files?

yuvaltassa, 2023-11-28:
I think there is some missing step somewhere in the instructions, you need to define some env variable?

Search the issues and let me know if you can't figure it out. Also I think we need to update the build instructions. You are not the first person to encounter this.

JeanElsner, 2023-11-28:
The simulate binary is looking for plugins in a folder called `mujoco_plugin` while your libraries are located in the `lib` directory after building. Copy the plugin `lib/libsdf.so` to `bin/mujoco_plugin` after building.

DimaRubyAI, 2023-11-29:
Thanks you @JeanElsner,
It's work for me with a small change, I'm using mac so in my case it's `libsdf.dylib` file and I was a bit confused because didn't find any folder in `bin` folder.. 
so in the bottom line:
after the build, in your build folder
1. create `mujoco_plugin` folder in `bin` folder
2. copy `libsdf.so` (windows?) or `libsdf.dylib` (macOs) file to `mujoco_plugin` folder

@nimrod-gileadi & @yuvaltassa, yes I think it's a good idea to add it to your tutorial, Thanks guys

closed 2023-11-29 by DimaRubyAI

RafaelsNeurons, 2024-03-14:
@DimaRubyAI @zichunxx @nimrod-gileadi  @yuvaltassa 
I dont find a libsdf.so file in windows, can somebody tell me where to find it?

A list of all steps to build from source would be nice:
the instructions on the website doesnt seem to be up to date
https://mujoco.readthedocs.io/en/stable/programming/index.html#building-from-source

DimaRubyAI, 2024-03-18:
@RafaelsNeurons sorry I never didn't with windows. On Mac the file is _libsdf.dylib_ and it located in _build/lib_ folder after successful build.