google-deepmind/mujoco #1239: [Unity] Issue collecting sensor data while using MPL hand

michaeljrose, 2023-11-28:
Hi,

I'm a student and I'm trying to use the MuJoCo Unity plug-in and HAPTIX to study human performance of referred robotic arm control. In particular, I am looking to have individuals use the MPL hand to interact with objects in an environment and complete tasks by moving said objects.

I'm looking for some help with creating sensorized objects in an environment with the MPL hand to see how strongly users are grasping the objects. In a new environment I have been able to make an object, create a site, add a MJ site scalar sensor (a touch sensor) and to get sensor values that make sense to me. See the first picture below, where I have a cylinder with mass = 1 in contact with a plane and giving a sensor reading of ~9.81 which makes sense due to gravity and it only reads that value once the cylinder has made contact with the plane (it initially starts above it and falls into the plane). 

When I import the MPL hand though, I start getting weird and even nonsensical values, even from the same sensorized object. In two examples below I have the same object in contact first with the same plane as when it previously worked, now giving a sensor value of 0.002, and in the second example the object is in contact with the table top imported with the MPL hand and it gives a value of -1.39 which doesn't make any sense as the scalar touch sensor should not be able to give negative sensor readings. Additionally, in these cases as soon as I enter play mode the sensor begins to read that value even if it is not in contact with anything in the environment. 

I was wondering if anyone could help me get to a point where the sensor values make sense when the MPL hand is around? 

Thanks,
Michael 

[image]
![image_720](https://github.com/google-deepmind/mujoco/assets/116675430/96bfb4a8-3
[image]
eba-4d9c-8873-cb19b49cafb0)

michaeljrose, 2023-11-29:
Hi sorry for not including versions, I am using mujoco 2.3.6 and Unity 2022.3.5f1

To display sensor readings I just edited the MjSiteScalarSensor script to make the SensorReading variable to be 
public double SensorReading;
and not 
public double SensorReading { get; private set; }

I don't believe I am using the hand in prefab mode, I brought it in by Assets --> Import MuJoCo Scene and then did not edit it at all, I don't think that is using it as a prefab but please let me know if that's mistaken. 

In terms of planes, I have still had the problem when I didn't add any planes to the MPL scene. 

My timestep is 0.002

Does anything stick out as to where I would be encountering a problem? Also, what version of Mujoco did you use where you weren't able to replicate the bug? 

Thanks

Balint-H, 2023-11-29:
Could you try to update to 3.0.1, reimport the hand, and add your extra cylinder and sensorize it?

michaeljrose, 2023-11-30:
It works as expected now! Thanks for suggesting, I didn't realize there was a new version of Mujoco out but that seems to have solved my problem.

Thanks again

closed 2023-11-30 by michaeljrose