google-deepmind/mujoco #1240: Tuning for higher simulation timestep

tawjaw, 2023-11-29:
Hi,

I'm an engineer and I'm trying to tune the assets I'm using with Mujoco to allow for consistent behaviour for different timesteps. 
My goal is to increase the timestep of the simulation for the environment I am using from 3ms to 10ms. 

I've set the solref of the geoms to -2500, -100 

With some of the meshes I am encountering a weird shaking behaviour. 

https://github.com/google-deepmind/mujoco/assets/10591561/af37db56-3020-4491-8770-9e4664885749

As seen in this video, the object is a convex hull and the ground is also a convex hull. 

My approach right now is to model the assets with primitives as much as I can. 

However, I would like to know is this behaviour expected at 10ms timestep? Are there tips I can follow to work with higher timesteps?

I couldn't find a reference on how to set the negative solref parameters for contact. What is a good default value for rigid bodies? 

Thanks

kevinzakka, 2023-11-30:
Hi @tawjaw, can you attach a minimal example (plane + mesh) and I can take a look.

yuvaltassa, 2024-01-02:
You can try using the [multiCCD](https://mujoco.readthedocs.io/en/latest/XMLreference.html#option-flag-multiccd) flag.

That said, 10ms is pretty large.

Closing due to lack of activity.

closed 2024-01-02 by yuvaltassa